MatMoul/g810-led #101: run as non-root

fjrclean, 2017-09-16:
I'm not sure as to the general conventions for device specific programs like this on linux, but wouldn't it be more secure to have g810-led run as it's own user? Couldn't you just make a logitech or g810 user on install; have the udev file give access to the devices to this user + run the program under this user; and make the systemd services run the program as this user. I have not looked at the source, but I assume this would work as the program only needs access to these devices, correct? You would prevent it from having complete root access which seems to me a bad security decision.

MatMoul, 2017-09-27:
You're right, it seem best to use a dedicated group.
But when a user is added to the group, the same problem appear, this allow a program to grab the keys...
It seem that only a driver can be resolve this problem.

hepan-git, 2018-12-01:
This is sort of related to the conversation of running as root: could someone maybe add something to the README or the command page that pops up when you run the command that tells you to run it as root. I spent a lot of time trying to figure out what was wrong and while searching I saw someone use sudo, so I did and it worked. It would just be nice for people to not make the same mistake I did, and if it actually says somewhere then please tell me.

hepan-git, 2018-12-26:
> 
> 
> g810-led does run without root and without setuid. It probably depends on your permissions but on most distros if the user is in the input group, it has access to the input devices, which is useful to launch a non-setuid Xorg btw.

I guess I'll try to add my user into the input group.

ShayBox, 2019-01-19:
Should so some checking for permissions, otherwise trying to use the program says no compatible device but its just because it needs to be ran as root

lanodan, 2019-01-20:
I guess you then have an issue with (e)udev or whatever is managing the dev files.

What does `ls /dev/input/` looks like on your system?

Example with mine, running eudev-3.2.5 (compatible udev 220):
```
ls -l /dev/input/
total 0
drwxr-xr-x 2 root root     100 Jan 16 03:03 by-id
drwxr-xr-x 2 root root     120 Jan 16 03:03 by-path
crw-rw---- 1 root input 13, 64 Dec 12 14:27 event0
crw-rw---- 1 root input 13, 65 Dec 12 14:27 event1
crw-rw---- 1 root input 13, 74 Dec 12 14:27 event10
crw-rw---- 1 root input 13, 75 Dec 12 14:27 event11
crw-rw---- 1 root input 13, 76 Dec 12 14:27 event12
crw-rw-rw- 1 root input 13, 77 Dec 12 14:27 event13
crw-rw---- 1 root input 13, 79 Dec 12 14:27 event15
crw-rw---- 1 root input 13, 66 Dec 12 14:27 event2
crw-rw---- 1 root input 13, 67 Dec 12 14:27 event3
crw-rw---- 1 root input 13, 68 Dec 12 14:27 event4
crw-rw---- 1 root input 13, 69 Dec 12 14:27 event5
crw-rw---- 1 root input 13, 70 Dec 12 14:27 event6
crw-rw---- 1 root input 13, 71 Dec 12 14:27 event7
crw-rw---- 1 root input 13, 72 Dec 12 14:27 event8
crw-rw---- 1 root input 13, 73 Dec 12 14:27 event9
crw-rw---- 1 root input 13, 63 Dec 12 14:27 mice
crw-rw---- 1 root input 13, 32 Dec 12 14:27 mouse0
```
`event13` is my keyboard btw, that can be found by looking in `by-id` or `by-path`.

muhamed89, 2019-02-16:
I have no idea but whats about flatpak? Will this work without root?

lanodan, 2019-02-16:
Pretty sure compiling g810-led statically would be much better than flatpak, but is there even systems without libusb-1.0?

But nah, shouldn’t require root if one could add joystick support (if it was already plugged) with just putting ``--device=all`` ( https://github.com/flatpak/flatpak/issues/7 ).

closed 2019-07-25 by MatMoul