MatMoul/g810-led #102: Color changing of more than one key but not a group in one command

Xeno72, 2017-10-05:
Hi MatMoul, first many thanks for your very cool and comfortable software! As a new owner of G810 and working 95 percent on linux, it is exactly what is needed to have full control over lighting effects of this keyboard.

I have a simple question: Is it possible to change the colour of more than one key (i. e. of all number keys) being not in a group in a single command? Something like:

`g810-led -k 1,2,3 00ff00`

to change the keys 1, 2 and 3 too green. Till now as I assume it has to be done in the following way 

```
g810-led -k 1 00ff00
g810-led -k 2 00ff00
g810-led -k 3 00ff00
```
This is no wrinkle if it is done in a script what I usually do, but to configure a higher number of single keys (not being a group!) manually it is a bit weary. Please note:

1) This is not a life and death decision.

2) **If there is a way to do similar by now, let me know.** I was very unsure to post this simple user's question here, but since no other way to contact you seems to exist (but you are perfectly entitled to handle it like that), I'm so bold.

Sincerely, X.

MatMoul, 2017-10-05:
You can make it with pipe command : 
echo -e "k 1 00ff00\nk 2 00ff00\nk 3 00ff00\nk 4 00ff00\nk 5 00ff00\nc" | g810-led -pp 

Don't miss the c for commit on the end of the line. 

Not very usefull but it's works...

Xeno72, 2017-10-05:
Dear MatMoul
That works perfectly! Since my know-how about the shell is limited I did not see this solution myself. Very good (and very fast!) answer. Chapeau!

In my view you're invited to close this issue.

Best wishes, X.

closed 2017-10-05 by MatMoul