MatMoul/g810-led #103: G502 support

n1zzo, 2017-10-16:
I'd like to add Logitech G502 Proteus Spectrum as a supported device,
can you give me a couple of guidelines on how to proceed?

gogosrcool, 2017-10-16:
The testing unsupported keyboards section of README.md can help you get started on the right foot.

MatMoul, 2017-10-16:
As @gogosrcool say, test with the unsupported section in README, if this not work, you need to make dump with Wireshark...
New mouses of Logitech seem to have the same microcontroller but protocol can differ.

n1zzo, 2017-10-17:
I tried all the three different protocols in the unsupported section of the README.
None of them work, now I'll be making a dump with wireshark.

MatMoul, 2017-10-24:
@n1zzo I try to read your dump but it's more difficult than I thought.
Give me some time, I'll work on it and provide you a first test method.
This issue can reactivate #40.

n1zzo, 2017-10-26:
Ok, maybe it's because the mouse has been used while it was tested.
If you want I can provide dumps where the mouse is not used as a pointing device (for example using another mouse).

MatMoul, 2017-10-26:
I've a new PR #106 from @Shulyaka that provide an initial support of the G403 mouse.
Now, from you dumps I've extracted these datas from set 00ff00 file :
11:ff:0e:7d:01:00:ff:00:00:00:00:00:00:00:00:01:00:ff:00:00
11:ff:02:3d:01:01:00:ff:00:02:00:00:00:00:00:00:00:00:00:00
11:ff:02:3d:00:01:00:ff:00:02:00:00:00:00:00:00:00:00:00:00

smasty, 2018-02-01:
Maybe #122  - Support for G203 Prodigy Mouse  can be of some help. I was able to decode the protocol used on that mouse.