MatMoul/g810-led #104: Compiling on FreeBSD

pyro3d, 2017-10-20:
Hi there,

I was able to get this running on FreeBSD 12 (technically TrueOS) with a couple modifications:

Changing all instances of u_int16_t in ./src/classes/Keyboard.h to uint16_t
In the makefile, changing the LDFLAGS from -lhidapi-hidraw to just -lhidapi

After that gmake bin worked and running the program as root allows me to change the intensity of the backlight on the G610 I have. 
Hope this is useful.

MatMoul, 2017-10-21:
Thanks for the info, I leave this issue open if anyone need it

plasmoduck, 2020-05-03:
I'm still having trouble building FreeBSD, I followed the instructions above. Using gmake says 

✗ gmake
c++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.4.0\"  -fPIC -shared -Wl,-soname,libg810-led.so -o lib/libg810-led.so.0.4.0 src/classes/Keyboard.cpp -lhidapi
ld: error: unable to find library -lhidapi
c++: error: linker command failed with exit code 1 (use -v to see invocation)
gmake: *** [makefile:42: lib/libg810-led.so] Error 1

plasmoduck, 2020-05-03:
Actually, after changing LIBS=-lhidapi to Lhidapi I got a little further, to this
✗ gmake
c++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.4.0\"  -fPIC -shared -Wl,-soname,libg810-led.so -o lib/libg810-led.so.0.4.0 src/classes/Keyboard.cpp -Lhidapi
c++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.4.0\"  src/main.cpp src/helpers/help.cpp src/helpers/utils.cpp src/classes/Keyboard.cpp -o bin/g810-led -Lhidapi
ld: error: undefined symbol: hid_close
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::~LedKeyboard())

ld: error: undefined symbol: hid_exit
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::~LedKeyboard())

ld: error: undefined symbol: hid_close
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::close())

ld: error: undefined symbol: hid_exit
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::close())

ld: error: undefined symbol: hid_init
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::listKeyboards())

ld: error: undefined symbol: hid_enumerate
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::listKeyboards())

ld: error: undefined symbol: hid_free_enumeration
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::listKeyboards())

ld: error: undefined symbol: hid_exit
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::listKeyboards())

ld: error: undefined symbol: hid_close
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_exit
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_init
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_enumerate
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_free_enumeration
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_exit
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_open
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_open
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_exit
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::open(unsigned short, unsigned short, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >))

ld: error: undefined symbol: hid_write
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::sendDataInternal(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >&))

ld: error: undefined symbol: hid_close
>>> referenced by Keyboard.cpp
>>>               /tmp/Keyboard-729cf1.o:(LedKeyboard::sendDataInternal(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >&))

ld: error: undefined symbol: hid_exit

lanodan, 2020-05-20:
> ld: error: unable to find library -lhidapi

Simple one: You do not have `hidapi` installed, `-L` (capital ell) is for adding directories to where it needs to find librairies (like `LD_LIBRARY_PATH` after it's linked).

lanodan, 2020-05-20:
Got it to compile here in a VM but couldn't test it against hardware. It's weird that passing `/usr/local/` paths needs to be done since it should be in the defaults already. And the `-lhidapi-hidraw` → `-lhidapi` is really annoying, I see no reason as to why it's different on FreeBSD other than maybe a lack of libusb and some legacy.

Anyway here is the diff
```diff
diff --git a/makefile b/makefile
index 8918b81..b4ba2b4 100644
--- a/makefile
+++ b/makefile
@@ -1,12 +1,13 @@
 CXX?=g++
 CXXFLAGS?=-Wall -O2
+PKGCONFIG?=pkg-config
 LIB?=hidapi
 ifeq ($(LIB),libusb)
        CPPFLAGS=-Dlibusb
        LIBS=-lusb-1.0
 else
-       CPPFLAGS=-Dhidapi
-       LIBS=-lhidapi-hidraw
+       CPPFLAGS=-Dhidapi -I/usr/local/include
+       LIBS=-L/usr/local/lib -lhidapi
 endif
 SYSTEMDDIR?=/usr/lib/systemd
```

plasmoduck, 2020-05-25:
Awesome! Thanks !

carpodaster, 2021-01-27:
Thanks for the work on this!

A gave @lanodan's patch a whirl on real hardware. Sadly it won't recognise my keyboard:

```
g810-led --list-keyboards
Matching or compatible device not found !Matching or compatible device not found ! 
```

Relevant `dmesg` output:

```
gen0.5: <Logitech Gaming Keyboard G810> at usbus0
ukbd0 on uhub0
ukbd0: <Logitech Gaming Keyboard G810, class 0/0, rev 2.00/2.14, addr 4> on usbus0
kbd2 at ukbd0
ukbd1 on uhub0

ukbd1: <Logitech Gaming Keyboard G810, class 0/0, rev 2.00/2.14, addr 4> on usbus0
kbd3 at ukbd1
```
(no idea why it's listing it twice :thinking: )

```
$ ldd `which g810-led`
/usr/local/bin/g810-led:
	libhidapi.so.0 => /usr/local/lib/libhidapi.so.0 (0x800270000)
	libc++.so.1 => /usr/lib/libc++.so.1 (0x80027a000)
	libcxxrt.so.1 => /lib/libcxxrt.so.1 (0x800347000)
	libm.so.5 => /lib/libm.so.5 (0x800369000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x80039b000)
	libc.so.7 => /lib/libc.so.7 (0x8003b5000)
	libusb.so.3 => /usr/lib/libusb.so.3 (0x8007ab000)
	libiconv.so.2 => /usr/local/lib/libiconv.so.2 (0x8007c4000)
	libthr.so.3 => /lib/libthr.so.3 (0x8008c4000)
```

```
$ pkg info -E hidapi
hidapi-0.8.0.r1_3
```

```
$ uname -vm
FreeBSD 12.2-RELEASE r366954 GENERIC  amd64
```

The keyboard is connected (i'm using it rn). Anything else I could supply to help with FreeBSD support? 

I modified the makefile to make it more FreeBSD'esque. Needs work still, but in case someone wants to pick it up: https://gist.github.com/carpodaster/b94bd054f27370dfc52088b07bb251d4

LastTrainToLondon, 2021-04-10:
https://reviews.freebsd.org/D29699