MatMoul/g810-led #105: G213 media buttons don't work

DableUTeeF, 2017-10-24:
I using G213 on ubuntu 16.04LTS, light setting works well but it cause media buttons(volume, play/pause) don't work.
They are fine if I unplug it and plug again.

MatMoul, 2017-10-24:
Not sure what's the problem because the code don't have action on multimedia keys.
With the udev rules, when you unplug and plug it again, g810-led is run.
Is your multimedia keys working the first time when you don't use g810-led?

retrofreak83, 2018-02-14:
Hey there,
I have the same problem, but with a G610. Media keys stopped working after installing g810-led on Ubuntu 16.04. But in my case the behaviour is stable, which means that unplugging and re-plugging the keyboard doesn't change anything. The LEDs are set to the initial default (function and meta keys on, rest off). Moreover, when I plug in the keyboard a bunch of error messages comes into syslog (see attached log).
[syslog.txt](https://github.com/MatMoul/g810-led/files/1723087/syslog.txt)

attatt, 2018-03-05:
Hi,
I have the same issue with a G810 on Ubuntu 16.04. It still occurs after restarts. Let me know if I can provide any more info. Thanks

MatMoul, 2018-03-06:
@retrofreak83 @attatt libusb is only supported for OS that don't support hidapi.
Use hidapi on Ubuntu, this will resolve your problem

retrofreak83, 2018-03-07:
That does the trick! Thanks!

closed 2018-08-01 by MatMoul