MatMoul/g810-led #109: Make install fails when DESTDIR is specified

iUnknwn, 2017-11-12:
I was attempting to install into /tmp/target so I could package this with fpm, and I noticed this problem:

    $ make install DESTDIR=/tmp/target
    cp: cannot stat '/etc/g810-led/samples/group_keys': No such file or directory
    makefile:83: recipe for target 'install' failed
    make: *** [install] Error 1

It looks like the makefile's post setup checks don't account for if DESTDIR was set.

MatMoul, 2017-11-12:
The command work great on ArchLinux.
What distribution do you use?

iUnknwn, 2017-11-12:
Fedora 25.

To me (though admittedly I haven't written a makefile in a while), it looks like the problem is here:
    
    @test -s /etc/$(PROGN)/profile || \
		cp /etc/$(PROGN)/samples/group_keys /etc/$(PROGN)/profile
	@test -s /etc/$(PROGN)/reboot || \
		cp /etc/$(PROGN)/samples/all_off /etc/$(PROGN)/reboot
	@udevadm control --reload-rules
    @$(PROGN) -p /etc/$(PROGN)/profile

All of the tests and copy operations are written against `/etc`, instead of `$(DESTDIR)/etc`

It also looks like the symlinks are potentially problematic in the `setup` routine too:

    @test -s $(DESTDIR)/usr/bin/g610-led || ln -s /usr/bin/$(PROGN) $(DESTDIR)/usr/bin/g610-led

Though this won't error, won't this attempt to generate a symlink pointing to `/usr/bin/$(PROGN)` instead of `$(DESTDIR)/bin/$(PROGN)`?

MatMoul, 2017-11-13:
I'll test with your distro...
@test /etc is volontary...
This is because when you make a package, you don't want to overwrite defined profile.

iUnknwn, 2018-02-16:
Any update on this issue?