MatMoul/g810-led #111: g610: Matching or compatible device not found error

0xMatt, 2017-12-06:
I'm not able to set anything due to this error. I checked --list-keyboards and it is indeed found.

```g610-led --list-keyboards
Device: Logitech - Gaming Keyboard G610
	Vendor ID: 046d
	Product ID: c338
	Serial Number: 0B8F36713135```

If any additional information is needed let me know and I'll get that for you, thanks.

MatMoul, 2017-12-09:
it seem you have found a new Product ID : c338.
If you want to make a PR, look at line 30 to 41 of src/classes/Keyboard.h

0xMatt, 2017-12-09:
Sorry, I meant to update this after I got it working. It seems all I needed was to reboot. After doing so it worked just fine. Thanks for your time!

closed 2017-12-09 by 0xMatt

danielamorais, 2019-06-09:
The same happened to me and worked after rebooting, thanks!

ID 046d:c338 Logitech, Inc. Gaming Keyboard G610
OS: Fedora 30