MatMoul/g810-led #113: Backlight button + number keys

Brynildsen, 2017-12-17:
I've noticed there are some functions while holding down the backlight button and pressing the number keys. I noticed them by coincidence pressing random buttons to see what happens. Some of those are accessible via the g810-led scripts here like cycle and wave effects, but especially "backlight button + 5" is cool, since it makes a wave outwards in all directions, from the specific key you press. Is something like that possible or could it be considered as a future enhancement? I'm working on some ideas to have it pick up beats from a musicplayer and flash different keys and that "number 5" would be absolutely awesome. :)

MatMoul, 2017-12-27:
When I've made dumps, I've not found data for this effect.

Brynildsen, 2017-12-27:
Thx anyway. How does one make dumps? I would like to learn.  :)

MatMoul, 2017-12-27:
Sorry, I don't explained why I can't dump data for this efffect.
When you set this effect with the Logitech software, after exit it, the effect stop working.
I can't dump data because the Logitech software don't use the internal effect,
I've tryed to send some data to the keyboard but at this time I've don't found a solution.

Brynildsen, 2017-12-28:
I stumbled upon this https://aurora.modworkshop.net/index.php from a post on reddit. It should have some functions that the original logitech windows software doesn't have, from what I read. They link to github at the site. Maybe there is some usable info there, they must have made dumps also somehow..

https://www.reddit.com/r/MechanicalKeyboards/comments/6c5797/how_to_get_a_ripple_color_effect_on_logitech_g810/

MatMoul, 2017-12-29:
@Brynildsen Thanks for your return, I'll try to look this project but g810-led is designed to work as a cmd and not as a daemon or a resident program.
To provide this type of effect, I need to make a keylogger and it's not the idea of g810-led.

closed 2018-09-13 by MatMoul