MatMoul/g810-led #114: Getting 'Matching or compatible device not found !' when keyboard is connected via USB hub.

radixs, 2017-12-31:
Hello guys,
I have a set-up where I flip my USB devices between two computers with the use of USB HUB. I have noticed that with my G610 your library works fine but only when the keyboard is plugged directly into the computer's USB port. When I plug it into the HUB they keyboard lights up only in the default mode (some areas are lit, but most like the letters are unlit) - which means the driver works but no fully (if it would not work I would have the keyboard's default wave effect).
When I do the --list-keyboards when connected via the USB hub I get the proper listing:
`Device: Logitech - Gaming Keyboard G610
        Vendor ID: 046d
        Product ID: c333
        Serial Number: 1078354C3033
`

but any other command gives me
`Matching or compatible device not found !`

any idea on how to fix this?

MatMoul, 2018-01-08:
Thanks for return, I never tested it on USB hub.
I'll try it when I found time

konserw, 2018-01-31:
I had the same issue with USB port on my motherboard, but after switching to another port it fixed itself.

michaelarnauts, 2018-10-05:
I see the same issue with my keyboard when it's connected trough a USB-C hub

nikklassen, 2018-10-23:
I also observed this behaviour when I connected my G610 to a USB-C port (no hub).

Galbi3000, 2019-10-20:
I get this problem too but my G910 keyboard is not connected through a USB hub.

I have tried using both hidapi and libusb methods.

I tried the --list-keyboards option and got this error: Segmentation fault (core dumped)

I am using Kubuntu 19.04 64-bit.

MatMoul, 2019-11-18:
@Galbi3000 Seem a build issue of g810-led, not an USB issue, open an other issue for this...

Galbi3000, 2019-11-19:
> @Galbi3000 Seem a build issue of g810-led, not an USB issue, open an other issue for this...

I'm sorry, I am not trying to be rude but from your response it seems to me you expected me to know the project well enough to know where I should post issues which is rather inconsiderate in my opinion. The general advice on GitHub is not to open new issues if there is a related issue already.

Consider it from my point of view: 

- I am tired of seeing the constant colour changing on my Logitech G910 Orion Spectrum keyboard as the shift from red to blue makes it look like the keys are moving and it bugs me while I am typing (I am not a touch typist!).
- g810-led was the ONLY software I could find that claimed to do what I wanted, that is take control of the keyboard LEDs just like can be done in Windows using Logitech's own drivers and stop the rainbow cycle effect.
- The README.md states clearly that g810-led works with my keyboard so I thought I should not have any problems.
- I followed the instructions in INSTALL.md to the letter and got no errors, this is normally a good sign. BTW you have "apt install g810-led" for Ubuntu installation but this should be "sudo apt install g810-led" to work properly.
- When I tried g810-led I was told "Matching or compatible device not found !".
- I followed the instructions to build the project from scratch (I.E. "git clone https://github.com/MatMoul/g810-led.git" etc.). 
- I tried running the executable and was again told "Matching or compatible device not found !".
- I read "hidapi is recommended but if you encounter a problem on your system, switch to libusb" so I did that with "make bin LIB-libusb". This also did not produce any build errors but still gave the same "Matching or compatible device not found !" error.
- I then tried running with the --list-keyboards option and got a segmentation fault.

As it builds without errors I had no reason to think it is a "build issue of g810-led" as you claim. So it very much does look like a USB issue to me from what I experienced as a new-comer to the project.

What I have not tried are the steps in the "Testing unsupported keyboards" section of README.md as I felt I should not have to do that when my exact keyboard is listed in the supported keyboards!

So with all that I did the what I thought was the right thing and searched for an existing issue that appears to match the one I am having before considering creating a new one. Apart from 3 differences (different keyboards, mine not in a USB hub and I get a segmentation fault with --list-keyboards) both this issue and my own appear to be related.

From my point of view this was the correct place for me to post and I have no indications to tell me otherwise. Instead of getting help or being asked for more information I got a curt response informing me I should not have posted here! Now you have more detailed information, can you tell me how I should have known I should have created a new issue?

MatMoul, 2019-11-19:
@Galbi3000 Sorry, I just try to maintain my issues and it's very hard...

The title of the issue end with "when keyboard is connected via USB hub" and it isn't your case.
The second reason is that you have Segmentation fault (core dumped)...
I'm pretty sure this isn't the same issue and this why I ask you to open a new one...

you're always welcome

ventorvar, 2020-01-25:
Having same issue with a Logitech G Pro keyboard. The USB hub is an active hub.

```
$ gpro-led --list-keyboards
Device: Logitech - Pro Gaming Keyboard
	Vendor ID: 046d
	Product ID: c339
	Serial Number: 178E36573635
$ gpro-led --print-device
Matching or compatible device not found !
```

```
$ lsubs -tv
[...]
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 0bda:5411 Realtek Semiconductor Corp. 
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:2812 VIA Labs, Inc. VL812 Hub
        |__ Port 4: Dev 9, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 2109:2812 VIA Labs, Inc. VL812 Hub
            |__ Port 4: Dev 10, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                ID 046d:0a5b Logitech, Inc. G933 Wireless Headset Dongle
            |__ Port 4: Dev 10, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                ID 046d:0a5b Logitech, Inc. G933 Wireless Headset Dongle
            |__ Port 4: Dev 10, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                ID 046d:0a5b Logitech, Inc. G933 Wireless Headset Dongle
            |__ Port 4: Dev 10, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                ID 046d:0a5b Logitech, Inc. G933 Wireless Headset Dongle
        |__ Port 2: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c339 Logitech, Inc. 
        |__ Port 2: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c339 Logitech, Inc. 
        |__ Port 3: Dev 7, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c53a Logitech, Inc. 
        |__ Port 3: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c53a Logitech, Inc. 
        |__ Port 3: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c53a Logitech, Inc. 

[...]
```

Galbi3000, 2020-01-25:
> Having same issue with a Logitech G Pro keyboard. The USB hub is an active hub.
> 
> ```
> $ gpro-led --list-keyboards
> Device: Logitech - Pro Gaming Keyboard
> 	Vendor ID: 046d
> 	Product ID: c339
> 	Serial Number: 178E36573635
> $ gpro-led --print-device
> Matching or compatible device not found !
> 
> ```

Try again with sudo, that worked for me.

MatMoul, 2020-05-24:
What the state if this issue?
sudo or not...
An other possibillity is that you use libusb.
Is it the case ?

muj-programmer, 2020-05-29:
Facing same issue if the keyboard is plugged into a usb hub.

kristoftorfs, 2020-06-05:
In my case using sudo fixed it (G513).

steven-jeanneret, 2020-06-24:
I had the same problem.
Then I restarted my computer (with keyboard plugged in the docking through USB-C) and now it works without sudo.

MatMoul, 2020-08-11:
@muj-programmer #114

filipbekic01, 2021-04-16:
Had same problem. Program must be run with `sudo` to avoid permission problems.

rlhelinski, 2021-08-06:
I had the same permissions issue. The program should check for write permissions to the USB device and report something like `No write permission for /dev/foo` rather than the misleading message `Matching or compatible device not found !`

eyaleb, 2021-09-04:
The program seems to not see the keyboard which is attached through a KVM. Probably has a built-in hub. I think this is the situation:

$ lsusb
Bus 001 Device 086: ID 0c45:7403 Microdia Foot Switch
Bus 001 Device 085: ID 1a40:0101 Terminus Technology Inc. Hub

$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 2: Dev 85, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 86, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 86, If 0, Class=Human Interface Device, Driver=usbhid, 12M

$ sudo bin/g810-led -dv 0c45 -dp 7403 -tuk 1 --list-keyboards
Device:  - 
        Vendor ID: 0c45
        Product ID: 7403
        Serial Number: 
Device:  - 
        Vendor ID: 0c45
        Product ID: 7403
        Serial Number: 

Same result with all protocols. Trying the '-a 000000' has no effect but I see the caps-lock light blinking briefly.
This is a G512 CARBON (GX BROWN TACTILE)

TIA

elvismdev, 2021-09-15:
`sudo g512-led -a 00ff00` worked for me.

eyaleb, 2021-09-17:
@elvismdev Was this a reply to my preceding note? If so, is your kbrd attached through a 1a40:0101 hub?

elvismdev, 2021-09-17:
> @elvismdev Was this a reply to my preceding note? If so, is your kbrd attached through a 1a40:0101 hub?

No, sorry no relation with your note, just posted for the whole thread that with sudo it worked for me.

ROBJkE, 2021-11-29:
had same issue. reboot solved it with no need of `sudo`