MatMoul/g810-led #115: G413 Matching or compatible device not found

umpirsky, 2018-01-05:
```
sudo apt-get install git g++ make libhidapi-dev
cd ..
git clone https://github.com/MatMoul/g810-led.git
cd g810-led
make bin
sudo make install

g413-led --list-keyboards
Device: Logitech - G413 Silver Mechanical Gaming Keyboard
   Vendor ID: 046d
   Product ID: c33a
   Serial Number: 017136643933

g413-led --print-device
Matching or compatible device not found !

cat /etc/issue
Ubuntu 17.10 \n \l

gnome-shell --version
GNOME Shell 3.26.1
```

I can provide more info if needed, I'm using wayland session. Thanks!

Brynildsen, 2018-01-05:
Did u reboot?
I constantly got it on my g810 at first when I installed, even though it registered fine on system.

umpirsky, 2018-01-05:
No, will try, and get back, thanks!

umpirsky, 2018-01-05:
Works, thanks!

closed 2018-01-05 by umpirsky

MatMoul, 2018-01-08:
@Brynildsen Thanks to help me to reply on issues.
Repplying to issues take more time than developping :(

cesar23, 2021-05-05:
`
> sudo apt-get install git g++ make libhidapi-dev
> cd g810-led
> make bin

  g++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.4.2\"  src/main.cpp src/helpers/utils.cpp src/helpers/help.cpp src/classes/Keyboard.cpp -o bin/g810-led -lhida

>sudo make install

> gpro-led --list-keyboards
Device: Logitech - PRO Gaming Keyboard
	Vendor ID: 046d
	Product ID: c339
	Serial Number: 0B8339543532
> gpro-led --print-device
Matching or compatible device not found !

> cat /etc/issue
Ubuntu 20.04.1 LTS \n \l

> gnome-shell --version
GNOME Shell 3.36.4

> reboot ....
`

reboot and it still persists the colors don't change. What I can be doing wrong

mpontes-stack, 2023-08-01:
In my Case I have to do (G213):

--
-- To resolve: "Matching or compatible device not found !" on G213 keybord in Ubuntu.
--

*) Install the package:

apt install g810-led

pip3 install PyGObject
pip3 install dbus-python
sudo apt-get install git g++ make libhidapi-dev

*) To solve the error about "Matching or compatible device not found !", recompile the sources:
   Based on post: https://github.com/MatMoul/g810-led/issues/115

- Git clone and make:

git clone https://github.com/MatMoul/g810-led.git
cd g810-led/
sudo make bin
sudo make install

 - Now the Keybord is recognize, the comands below:

g213-led --list-keyboards
g213-led --print-device
sudo g213-led --print-device

 - shows: 

Device: Logitech - Gaming Keyboard G213
        Vendor ID: 046d
        Product ID: c336
        Serial Number: 136A30533330

--- !!! than it's work !
*) Simple test:
   g213-led -a ff0000

*) A good post for configuring g213 logitech keyboard.

https://www.electricbrain.com.au/pages/linux-rgb-keyboard.php

*) With the post I configued my own profile, my profile now :

# Login profile by groups keys

r 1 33FFF6
r 2 33FFF6 
r 3 33FFF6
r 4 FFFFFF
r 5 CA6F1E

c # Commit changes

*) than I create a file: g213_profile.dat in /etc

To load this on boot time I add on my root crontab the line:

@reboot sleep 10 && /usr/bin/g213-led -p /etc/g213_profile.dat