MatMoul/g810-led #116: LEDs auto-off when not in use

tobru, 2018-01-07:
Is there a possibility to turn of the LEDs after a few minutes of inactivity and turn them on again as soon as a key is pressed?

Brynildsen, 2018-01-07:
You could make a profile with all LEDS off. And easily call that profile with a script or a cron job of some sort, that will run itself after a predefined amount of inactivity. Just make a script yourself and learn howto. If you ask in a forum or one of the reddit linux subs you will get a quick reply from someone who knows. Basicly it should work like a screensaver, but just run another process. Don't forget to make it call you LED profile again on exit, so they will turn on again.

MatMoul, 2018-01-08:
G810-led is not a daemon, it can't track state of the computer and I don't want to make a keylogger.
The @Brynildsen's reply offer a first solution based on the screensaver.
But if you want a custom result, build g810-led as library and make your own binary that can track keypress. Use a timer to set all keys off and set your profile again.

lanodan, 2018-08-07:
It’s a bit different from what you want but I currently load a profile when my screen locks and it reload the default profile when it unlocks.

Done it this way (I use ``slock`` for locking, replace it with the command to lock/screensave):
```
# Relevant stuff from my .xinitrc
g810-led -pp < /home/haelwenn/.config/g810-led-profile &
xset +dpms dpms 40 60 200 s 40 s expose s on &
xautolock -time 0 -locker locker -secure -detectsleep &
```
```
#!/bin/sh
# Copy of ~/.local/bin/locker
g810-led -a 00 ; g810-led -fx breathing logo 7f 0a
slock
g810-led -pp < /home/haelwenn/.config/g810-led-profile
```

closed 2018-09-13 by MatMoul