MatMoul/g810-led #117: [US version] backslash \ and pipe | keys not recognized in the python version

makomoulage, 2018-01-16:
Hi MatMoul,

Thanks again for your hard work on this tool.
I have to use the python version as I'm using this keyboard at work, where I don't have persmissions to do anything and people are reluctant to customize Centos on a per user basis.

The issue I've got is that the US layout description is different from the one you might be using (French I guess?). For example, the question mark (?) is called "Minus" in the sampleprofile file.

But more importantly, I can't get the backslash key to change color.
There is a "backslash" thing in the file, but it's color is kept unchanged.
By any chance, do you know how to fix that ?
Digging in your python script, it's calling the key:64.

Thank you again

Brynildsen, 2018-01-17:
Does it help if you "escape" those special characters in your commands?

makomoulage, 2018-01-17:
Here is what I've tried so far, in my sampleprofile file:
backslash = 'ff0000'
intl_backslash = 'ff0000'
unknown = 'ff0000'
Asterisk = 'ff0000'
NumEqual = 'ff0000'

None of them changed my '\' (and '|') to red.

MatMoul, 2018-01-17:
Hi, the python code is unmaintained from time but I've take a look on it and at line 168-169, key 31 is missing.
Perhaps it's the problem...
try to add a line like this :
elif (key == 'mybackslash') : key = '31'
and use mybackslash as key name...

makomoulage, 2018-01-17:
Excellent ! This now works !
You're a genius !
Thanks a lot for your help !

closed 2018-01-17 by MatMoul