MatMoul/g810-led #119: Constantly running/logging. Possible keyboard malfunction over time?

kenwardv, 2018-01-21:
Thank you for providing this to those who have this keyboard (mine's an 810). 

Have been running for a month. Just noticed the other day that the system constantly logs all action of the keyboard, and shows failures. Example syslog:

...
Jan 21 10:48:47 hpotter systemd-udevd[8182]: Process '/usr/bin/g810-led -p /etc/g810-led/profile' failed with exit code 2.
Jan 21 10:48:47 hpotter systemd-udevd[8181]: Process '/usr/bin/g810-led -p /etc/g810-led/profile' failed with exit code 2.
Jan 21 10:48:47 hpotter kernel: [ 2308.107268] input: Logitech Gaming Keyboard G810 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.1/0003:046D:C337.15D4/input/input5597
Jan 21 10:48:47 hpotter kernel: [ 2308.164990] hid-generic 0003:046D:C337.15D4: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G810] on usb-0000:00:1d.0-1.2/input1
Jan 21 10:48:48 hpotter systemd-udevd[8229]: Process '/usr/bin/g810-led -p /etc/g810-led/profile' failed with exit code 2.
Jan 21 10:48:48 hpotter systemd-udevd[8181]: Process '/usr/bin/g810-led -p /etc/g810-led/profile' failed with exit code 2.
Jan 21 10:48:48 hpotter kernel: [ 2308.294133] input: Logitech Gaming Keyboard G810 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.1/0003:046D:C337.15D5/input/input5598
Jan 21 10:48:48 hpotter kernel: [ 2308.353117] hid-generic 0003:046D:C337.15D5: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G810] on usb-0000:00:1d.0-1.2/input1
...

This is filling my log files.  The one way I found to stop it was to uninstall g810-led after allowing it to setup the keyboard.  

How can I do that without such a kludge?

Today I found the keyboard in a state where it appeared to be locked on a certain input (e.g. periods, or "p]" being pressed constantly.  Rebooting brought back the same state.  Didn't play enough with it to see whether unplugging or other actions could help, but did note that rebooting into another system did not stop the repeating character. 

System:  Debian Stretch, completely up to date.  Compiled using libusb.  

What other info should I provide?

kenwardv, 2018-01-21:
Hmmm.  Wow.  Just recompiled using hidapi and all the logging, etc, seems to have stopped.  

Obviously don't know about the "stuck key" issue yet.  Will keep an eye out for that one.

MatMoul, 2018-01-22:
I don't have idea for your log, hidapi is recomended.
But for your "stuck key", g810-led is a one shot cmd... and it work on the second usb endpoint of the keyboard, this can't affect keypress.

kenwardv, 2018-01-22:
Thanks for getting back to me.   As I mentioned in the second post, the hidapi made the difference for me.  I'm just wondering what may be going on in the system between the keyboard, its interface, and something else which may have intruded on that to generate a signal like that.  Frustrating.

Again, I appreciate what you did!  

One final thought would be the inclusion of a line like "Try hidapi first--libusb has been known to create problems of its own."

Cheerio!

closed 2018-01-22 by kenwardv

MatMoul, 2018-01-27:
Thanks for your return...
I just rereaded the https://github.com/MatMoul/g810-led/blob/master/INSTALL.md file and I have a section (hidapi vs libusb) that seem explicit... or not ?