MatMoul/g810-led #125: Load profile on recovery from standby?

retrofreak83, 2018-03-08:
Hi there,
I have a G610 and running Ubuntu 16.04. When connecting the keyboard when the system is running, the profile is loaded automatically. When recovering from standby, the keyboard is initialized with what is defined via "g810-led --startup-mode". Is there a possibility to automatically load my profile under /etc/g810-led/profile?

MatMoul, 2018-03-21:
can you be more specific for "Is there a possibility to automatically load my profile under /etc/g810-led/profile?"

retrofreak83, 2018-03-22:
When I connect the keyboard when the system is running or I dock the laptop to the docking station (where the keyboard is connected), the profile located under /etc/g810-led/profile is loaded automatically.
This is not the case when I put the laptop with the keyboard connected in standby. When the laptop comes up from standby again, not the profile is loaded but it comes up with the mode given with g810-led --startup-mode, so I can only choose between the standard wave effect or all keys glowing.

When I do a g810-led -p /etc/g810-led/profile manually or disconnect and reconnect the keyboard from/to the computer, everything is fine again.

MatMoul, 2018-04-08:
I don't have studied this case.
On recent system with systemd, make a service unit for this.
ex : https://unix.stackexchange.com/questions/124212/writing-a-systemd-service-to-be-executed-at-resume

FiSHYsolutions, 2021-03-05:
i had the same issue using a G512 with xubuntu 20.04.2 LTS and g810-led v0.3.9-1build1
adding a unit-file solved my problem:

```
bash$ cat /usr/lib/systemd/system/g810-led-wakeup.service
[Unit]
Description=set G512 profile on resume after standby
After=suspend.target

[Service]
Type=oneshot
TimeoutSec=0
StandardOutput=syslog
ExecStart=/usr/bin/g810-led -p /etc/g810-led/profile

[Install]
WantedBy=suspend.target
```
```
bash$ sudo systemctl enable g810-led-wakeup.service
Created symlink /etc/systemd/system/suspend.target.wants/g810-led-wakeup.service → /lib/systemd/system/g810-led-wakeup.service.
```