MatMoul/g810-led #126: .0--

Silver-Shroud, 2018-03-12:


MatMoul, 2018-03-21:
Give me the result of the next command :
`lsusb`

Silver-Shroud, 2018-03-21:
lsusb
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 011 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 010 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 003: ID 045e:028e Microsoft Corp. Xbox360 Controller
Bus 009 Device 002: ID 0b05:181a ASUSTek Computer, Inc. 
Bus 009 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 002: ID 046d:c337 Logitech, Inc. 
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

MatMoul, 2018-04-08:
Your product is reported as Logitech 046d:c337 that is supported by g810-led.

Silver-Shroud, 2018-04-08:
These keys are not supported. They are only lit up dark purple because I rebooted to Windows and back.
[image]

MatMoul, 2018-04-08:
I need a dump with wireshark when you set the profile with windows or a sdk that contain these keys...

Silver-Shroud, 2018-04-08:
OK, I'll do that later today. Thanks.

Silver-Shroud, 2018-04-09:
[G810-Capture.pcapng.zip](https://github.com/MatMoul/g810-led/files/1888534/G810-Capture.pcapng.zip)

I made a lighting zone for those keys specifically and played around with the colours. I hope that helps.

MatMoul, 2018-04-11:
Ok it seem that i've found keys in the dump :
11:ff:0c:1b:00:01:00:38:00:00:00:00:00:00:00:00:00:00:00:00
11:ff:0c:2b:00:01:00:38:00:00:00:00:00:00:00:00:00:00:00:00
11:ff:0c:3b:00:02:00:05:00:00:00:00:00:00:00:00:00:00:00:00
11:ff:0c:4b:00:02:01:00:00:00:00:00:00:00:00:00:00:00:00:00
11:ff:0c:5b:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

But can you give me an english name for each key and if it is a char, a modifier or other...

Silver-Shroud, 2018-05-26:
Is that information sufficient?

MatMoul, 2018-06-11:
Yes thank you but I'm verry busy this year...
I'll tried to work on it as soon as possible

Silver-Shroud, 2018-06-11:
No problem. I'll use Windows to configure those keys for now.

nerokeygen, 2020-04-16:
Hi! I just came here now searching how to change color of these unsupported keys.
in case you didn't know, I figured out that I can use g810-led -fx color all (RGB CODE) to fix colors for all keys. all other effects also works.

Hayao0819, 2020-08-12:
I would like G512 to provide additional key color changes for Japanese as well.
I will provide as much information as I can.
The picture below is the result of executing the `g910-led -a 000000` command.

[image]

Hayao0819, 2020-08-12:
Here are the results of staring at lsusb:
```
Bus 003 Device 006: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 008: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 007: ID 046d:c246 Logitech, Inc. Gaming Mouse G300
Bus 003 Device 009: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 003 Device 005: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 046d:c33c Logitech, Inc. 
Bus 001 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```