MatMoul/g810-led #128: Add clang support in makefile

talybin, 2018-04-05:
Hi, your makefile is hardcoded to g++ however in most distros there is a link pointing to currently used compiler, c++. Also clang gives an error compiling header files which are safe to remove. My suggestion to changes is following.
```
diff --git a/makefile b/makefile
index 019636d..7c3b572 100644
--- a/makefile
+++ b/makefile
@@ -1,5 +1,5 @@
-CC=g++
-CFLAGS=-Wall -O2 -std=gnu++11
+CC=c++
+CFLAGS=-Wall -O2 -std=c++11
 LIB?=hidapi
 ifeq ($(LIB),libusb)
        CPPFLAGS=-Dlibusb
@@ -21,8 +21,8 @@ MINOR=2
 MICRO=7
 
 CFLAGS+=-DVERSION=\"$(MAJOR).$(MINOR).$(MICRO)\"
-APPSRCS=src/main.cpp src/helpers/*.cpp src/helpers/*.h
-LIBSRCS=src/classes/*.cpp src/classes/*.h
+APPSRCS=src/main.cpp src/helpers/*.cpp
+LIBSRCS=src/classes/*.cpp
 
 .PHONY: all bin debug clean setup install uninstall lib install-lib install-dev
```

MatMoul, 2018-04-08:
I'll review #121 and after that I'll comming back to this issue

talybin, 2018-04-20:
If you choose to use cmake, compiler will be selected automatically. But still you will need to get rid of header files in list of sources.

MatMoul, 2019-01-04:
Closing : https://github.com/MatMoul/g810-led/pull/121#issuecomment-449885163

closed 2019-01-04 by MatMoul