MatMoul/g810-led #131: G513 Carbon support

derMalte, 2018-05-09:
Adding support for the G513 Carbon would be greatly appreciated. 

I've tried testing the unsupported keyboards section of README.md, but without success.

I'm fairly new to this, but if required I can try to create a wireshark dump. Any guidance on how to do this?

Thanks,
derMalte

MatMoul, 2018-06-05:
Hello and thank you to present a new keyboard...
Firstly, provide your product ID, you can find it with lsusb...
if you have already tested the unsupported method, it seem a wireshark dump is needed...
I'm missing time but I'll try to reply to you...

danpascu, 2018-06-06:
The keyboard is mostly supported through protocol 1. This command works on it:

g810-led -dv 046d -dp c33c -tuk 1 ...

From what I tested everything seems to work, except setting the logo (it will not light up at all) and the breathing effect with a given color.

G513 is their top of the line keyboard, so it may even support more effects than G810, but I haven't checked their windows software to see what it can do. The keyboard definitely has a logo, so the protocol for that may have changed, as for the breathing effect I'm not sure if it's not supported or it just changed the protocol for that too.

MatMoul, 2018-06-08:
@danpascu Thanks for return, this is a good information...

derMalte, 2018-06-10:
Hi! 

Thanks for the help @danpascu. As far as I can tell from the looks of my keyboard and the  [product page](https://www.logitechg.com/en-us/product/g513-backlit-mechanical-gaming-keyboard), I don't think the logo has any lighting, so it seems that the keyboard is fully supported through protocol 1 for you. 

Unfortunately I can't get it to work on my machine yet:

```
$ lsusb
Bus 005 Device 002: ID 046d:c33c Logitech, Inc.
```

```
$ g810-led -dv 046d -dp c33c tuk 1 -a 00ff00
Matching or compatible device not found !
$ g810-led -dv 046d -dp c33c tuk 2 -a 00ff00
Matching or compatible device not found !
$ g810-led -dv 046d -dp c33c tuk 3 -a 00ff00
Matching or compatible device not found !
```

Is there anything obvious I might have done wrong?

Thanks,
Malte

danpascu, 2018-06-11:
Run the command as root (or use sudo). You do not have the rights the access the raw hid device as an unprivileged user.

derMalte, 2018-06-11:
Hi,

tried it with sudo as well:

```
$ sudo g810-led -dv 046d -dp c33c tuk 1 -a 00ff00
Matching or compatible device not found !
$ sudo g810-led -dv 046d -dp c33c tuk 2 -a 00ff00
Matching or compatible device not found !
$ sudo g810-led -dv 046d -dp c33c tuk 3 -a 00ff00
Matching or compatible device not found !
```

Any other thoughts?

Appreciate your help!

Best regards,
Malte

MatMoul, 2018-06-11:
@derMalte the tuk parameter need to be writed -tuk..
`sudo g810-led -dv 046d -dp c33c -tuk 1 -a 00ff00`

MatMoul, 2018-06-11:
@danpascu If protocol 1 work, it's a good start...
Now for the logo, if you can, try to make a dump...
For the other effects, you can test it easily with starting it with the Windows software and exit it.
If the effect still working, it is supported by the keyboard...

rthrmt, 2018-06-12:
I've used this keyboard yesterday on Windows. 
I can confirm that there isn't a LED for the logo, so it can't light up anyway.

The effects seem to be the same as on Windows, I can give confirmation about this later this day.

MatMoul, 2018-06-17:
@rthrmt Thank for your return, it seem that I can add this model with protocol 1...

derMalte, 2018-06-20:
@MatMoul I can confirm that (once I got the syntax right) protocol 1 works. Thanks for the patience and your work!

MatMoul, 2018-07-09:
No, native fx are related to the entire keyboard

hsegnitz, 2018-07-25:
FYI: Just to let you know, I've received an hour ago my shiny new G512 keyboard (not a typo, it's a new one in their lineup - basically 513 without wristrest) and it has the same hardware IDs and is also working with protocol 1.

```Bus 001 Device 013: ID 046d:c33c Logitech, Inc. ```

```sudo g810-led -dv 046d -dp c33c -tuk 1 -a 000000```

MatMoul, 2018-07-26:
@hsegnitz Thanks for your return, I need to work on these two new keyboards...

MatMoul, 2018-08-01:
Try to build this branch : https://github.com/MatMoul/g810-led/tree/g513
This is the first support of G512 and G513...
I wait return brefore merging...

scones, 2018-08-08:
i can confirm this branch to be working on my g513.

although the new effects are missing.
but i won't complain :)

thanks for your effort!

MatMoul, 2018-08-08:
@scones Thanks for your return, I can now merge the change...

For the new effects, I'm can't test it...
On g810 some effects are integrated in the keyboard and I need a dump to integrate it...
If you won to test if the new effects are native, launch the effect on Windows and exit the Logitech software. If effects stop working, it's not native effects and I need a dump to add it to g810-led...

scones, 2018-08-09:
As i currently have this keyboard only on my linux machine, i can't test this.
I will wait and see if someone else can provide information

scones, 2018-08-16:
Out of curiosity: How to generate a dump?

MatMoul, 2018-09-13:
G512 and G513 is now published in the new release...
@scones, I've searched my previous post that explain the method but I don't find it...
I'll try to make a wiki page for this...

closed 2018-09-13 by MatMoul