MatMoul/g810-led #133: Device not found

johnmanko, 2018-05-31:
List the device
```
$ ./bin/g810-led --print-device
Matching or compatible device not found !
```

List all keyboards
```
$ ./bin/g810-led --list-keyboards
Device: Logitech - Gaming Keyboard G810
	Vendor ID: 046d
	Product ID: c337
	Serial Number: xxxxxxxxxxxx

```

Set the color:
```
$ ./bin/g810-led -a 00ff00
Matching or compatible device not found !
```

minimec, 2018-06-01:
Looks like you didn't (or don't want to) install the software 'system wide' with 'sudo make install'. In that case you are missing the udev rule file in /etc/udev/rules.d, that gives the user access to the device. You find that file in the 'udev' directory in the build folder.

In case my suggestions are correct you should at least do the following command on your system to add a rule for your specific device.

```
echo 'ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c337"' | sudo tee -a /etc/udev/rules.d/g810-led.rules

sudo udevadm control --reload-rules && udevadm trigger
```

That should do the trick

johnmanko, 2018-06-02:
@minimec Thanks.  I'll try later.

MatMoul, 2018-06-05:
I confirm, as user you need to use udev rules

johnmanko, 2018-06-05:
Ok. I guess you can close this.

closed 2018-06-08 by MatMoul