MatMoul/g810-led #136: g810: sudo make install fails

Dobeedoo, 2018-06-30:
Hi,
been using this for some time now, but had to reinstall today. It builds ok, but 'sudo make install' fails with:
$ sudo make install
make: execvp: g810-led: Permission denied
makefile:83: recipe for target 'install' failed
make: *** [install] Error 127

Any idea? Running Debian 9.4 (stretch) and Gnome desktop. 
Thanks!

MatMoul, 2018-07-03:
Showing the $, it's seem that you are logged as user and it's good.
But is the user a sudoer user ?

Dobeedoo, 2018-07-07:
Hi, yes the user has full sudo permissions. I have also tried to do this as "real" root as well (#), but same problem;

# make bin LIB=libusb
g++ -Dlibusb -Wall -O2 -std=gnu++11 -DVERSION=\"0.2.7\" src/main.cpp src/helpers/utils.cpp src/helpers/help.cpp src/helpers/help.h src/helpers/utils.h src/classes/Keyboard.cpp src/classes/Keyboard.h -o bin/g810-led -lusb-1.0
# make install
make: execvp: g810-led: Permission denied
makefile:83: recipe for target 'install' failed
make: *** [install] Error 127
#

I have of course installed the dependencies mentioned in INSTALL.md. Thinking some package still might be missing...?

Thanks

MatMoul, 2018-07-09:
It seem that you have a permission problem...
Try to login as user, make a clean git clone...
Go to the git directories and run :
make bin LIB=libusb # But hidapi lib is recommended
After that :
sudo make install

Dobeedoo, 2018-07-09:
Same thing I'm afraid. I already did this once, but re-did it now again once more as you asked me to;
$ git clone https://github.com/MatMoul/g810-led.git
Cloning into 'g810-led'...
remote: Counting objects: 1405, done.
remote: Total 1405 (delta 0), reused 0 (delta 0), pack-reused 1405
Receiving objects: 100% (1405/1405), 819.13 KiB | 808.00 KiB/s, done.
Resolving deltas: 100% (676/676), done.
$ cd g810-led/
$ make bin LIB=libusb
g++ -Dlibusb -Wall -O2 -std=gnu++11 -DVERSION=\"0.2.7\" src/main.cpp src/helpers/utils.cpp src/helpers/help.cpp src/helpers/help.h src/helpers/utils.h src/classes/Keyboard.cpp src/classes/Keyboard.h -o bin/g810-led -lusb-1.0
$ sudo make install
make: execvp: g810-led: Permission denied
makefile:83: recipe for target 'install' failed
make: *** [install] Error 127
$
Have others reported this to work on Debian Stretch (9.4) ? My Debian install is all stock, no specials and definitively no change in default permissions...

Dobeedoo, 2018-07-09:
This was really weird. Just for fun I wanted to see if "sudo make uninstall" worked. It did not give an error, so expect it worked. Then tried "sudo make install" again...;
$ sudo make install
Created symlink /etc/systemd/system/shutdown.target.wants/g810-led-reboot.service → /usr/lib/systemd/system/g810-led-reboot.service.

Folders got created as they should. All it took was apparently an uninstall before the actual install.

Thanks

MatMoul, 2018-07-09:
@Dobeedoo Thank's for your return, I'll try to sea if I can handle this case...

Dobeedoo, 2018-07-14:
I made a few more interesting findings. On Debian 9.4 (stretch), the lib "libusb-1.0-0-dev" breaks keyboard multimedia controls, such as volume etc. I had problem with that suddenly stopped working, and could not find a way repair it. So I eventually reinstalled, testing thoroughly after each installed application, and after having installed above lib, it broke. Uninstalling it made the controls work again. Maybe worth some additional testing and a note in the INSTALL.md ?
After last reinstall, 'sudo make install' worked directly, so tempted to think previous problems may have been a quirk in Debian...

MatMoul, 2018-07-16:
It's possible that libusb will cause you some problems...
This is why I recommend hidapi (https://github.com/MatMoul/g810-led/blob/master/INSTALL.md)...
Why do you use libusb ???

tekwon, 2018-09-13:
I had problems installing the g513 branch on Ubuntu 18.04.

Was getting error:
```
cp: cannot stat 'bin/g810-led': No such file or directory
makefile:56: recipe for target 'setup' failed
make: *** [setup] Error 1
```

Running:` sudo make debug`
then:`sudo make install `
fixed the problem.

MatMoul, 2018-09-13:
Very strange but thanks for the return...
I close the issue

closed 2018-09-13 by MatMoul