MatMoul/g810-led #137: Please include copyright notices

mgorny, 2018-08-17:
It seems that the code files lack appropriate copyright notices. Please include them to make it clear what the license is (especially that GPL is used in two variants, 'GPL version 3 *only*' and 'GPL version 3 or newer'. See the snippet on bottom of LICENSE.

MatMoul, 2018-08-27:
I just used the default license provided by Github...
I'll try to learn the diff between the two license...

MatMoul, 2019-01-04:
I'm not sure what I need to make...
Adding some comment on top of each source file ?
But I don't understand "GPL version 3 only' and 'GPL version 3 or newer"

mgorny, 2019-01-04:
Yes, you basically should include a copyright notice on top of each copyrightable file. See https://www.gnu.org/licenses/gpl-howto.html.en; particularly 'license notices' section.

The 'only'/'or newer' is basically a upgrade clause. If you state 'version 3 or newer', it means that when FSF release GPL-4, users will be permitted to immediately use its terms instead of the terms of version 3. If you make it 'version 3 only', users won't be permitted to use GPL-4 until you explicitly state it's allowed.

MatMoul, 2019-01-05:
@mgorny I found just now...

MatMoul, 2019-01-05:
@mgorny Thanks for the training...
I've made a commit for this on the develop branch...
If you have remarks...

MatMoul, 2019-01-15:
Release 0.3.2 include now notices changes

closed 2019-01-15 by MatMoul