MatMoul/g810-led #138: G512 LIGHTSYNC RGB Mechanical Gaming Keyboard support

menon92, 2018-08-21:
Does this keyboard support RGB color in Linux machine ? If not then How can I do it ? or logitech have any RGB color keyboard that support in Linux machine.

Thanks in advance @MatMoul

pearsonk, 2018-08-21:
I did some cursory tests and the G512 seems identical to the G513 (same vendor ID, device ID, and protocol bits). If you run tests with the G513 integration branch and provide feedback, that would be great. I recently moved and am still waiting on my Logitech devices to be shipped.

MatMoul, 2018-08-27:
Following this issue #131 , @hsegnitz has say that the G512 is the same as G513...
You can test the G513 branch and return me if that work...
I'll merge this branch soon...

PS: Thank's to @pearsonk to help me in the comment and in the code...

closed 2018-09-13 by MatMoul

odziom91, 2019-11-30:
Hello,
I bought G512 Special Edition keyboard. After installation g810-led-git (package from AUR in ArchLinux) and running with parameters --list-keyboards and --print-device I received information:
Matching or compatible device not found !
and the same info with sudo.
I tried to testing my keyboard. I put lsusb: 046d:c342 and test keyboard with command:
g512-led -dv 046d -dp c342 -tuk 1 -a FF0000
Keyboard changed color to red so it is ok. How to make this keyboard working without additional -dv -dp and -tuk params?

Thanks in advance!