MatMoul/g810-led #140: How to use this file. Can someone give an example?

habj89, 2018-09-08:
Hello
New Linux Ubuntu user here.
I have managed to install this, and the rainbow effect stopped on my g910 keyboard.
But how do I actually change the color?
I understand that I have to get into the terminal, and change it from there, but don't know how to.
`g910-led --help`
Then I get this:
`g910-led
Version : 0.2.7

Usage: g910-led [OPTIONS...] [command] (command arguments)
Commands:
  -a {color}				Set all keys color
  -g {keygroup} {color}			Set key group color
  -k {key} {color}			Set key color
  -mr {value}				Set MR key (0-1)
  -mn {value}				Set MN key (0-7) (M1=1, M2=2, M3=4) (M1+M2=3, M1+M3=5, ...)

  -gkm {value}				Set GKeys mode (0=Mapped to FKeys, 1=Independent)

  -an {color}				Set all keys color without commit
  -gn {keygroup} {color}		Set key group color without commit
  -kn {key} {color}			Set key color without commit
  -c					Commit change

  -fx ...				Use --help-effects for more detail

  < {profile}				Set a profile from a file (use --help-samples for more detail)
  |					Set a profile from stdin (for scripting) (use --help-samples for more detail)

  --startup-mode {startup mode}		Set startup mode

  --list-keyboards 			List connected keyboards
  --print-device			Print device information for the keyboard

  --help				This help
  --help-keys				Help for keys in groups
  --help-effects			Help for native effects
  --help-samples			Usage samples

Options:
  -dv					Device vendor ID, such as 046d for Logitech. Can be omitted to match any vendor ID
  -dp					Device product ID, such as c337 for Logitech G810. Can be omitted to match any product ID
  -ds					Device serial number, Can be omitted to match the first device found
  -tuk					Test unsuported keyboard with one of supported protocol (1-3) -dv and -dp are required

Values:
  color formats :			RRGGBB (hex value for red, green and blue)
  speed formats :			SS (hex value for speed 01 to ff)

  key values :				abc... 123... and other (use --help-keys for more detail)
  group values :			logo, indicators, fkeys, ... (use --help-keys for more detail)
  startup mode :			wave, color
`
I fail at writing the right line to change the lights.

MatMoul, 2018-09-11:
Not sure if this is what you want :
`g910-led -a 0000ff`

habj89, 2018-09-11:
Thank you for replying, but when I run
`g910-led -a 0000ff`
I get:
Matching or compatible device not found !

My keyboard is a G910 Orion Spectrum. 
I think it is installed right because after the install the wave effect stopped. 
And when I type:
`g910-led --help`
I get the whole menu up in the terminal.
Is there some place else I messed up?

MatMoul, 2018-09-12:
Firstly try the next command to list supported keyboard on your system :
`g910-led --list-keyboards`
If your keyboard is listed, it is supported...
If it is listed and your receive :
Matching or compatible device not found !
Try with sudo, this will work...
For using as user, the magic file is https://github.com/MatMoul/g810-led/blob/master/udev/g810-led.rules

habj89, 2018-09-12:
Thank you so much for taking your time to help me.
All that was needed was sudo.
Should have thought to try with with that earlier on.

closed 2018-09-12 by habj89