MatMoul/g810-led #142: G512/G513 Partially functional

pearsonk, 2018-09-18:
As of tag v0.2.8 (https://github.com/MatMoul/g810-led/commit/b707deb6d3eee419aaa0d27db7b0a538ee4d835a), the G512 and G513 (0x046d:0xc33c) are only partially functional with this. 

The native effects are functional, but setting individual keys is not.

pearsonk, 2018-09-18:
[g512-filtered.zip](https://github.com/MatMoul/g810-led/files/2392813/g512-filtered.zip)
Attached a wireshark dump where I simply set the number keys (n1-n0) to red (0xFF0102). See packet 482 of the attachment. It seems here that G Hub is only setting the keys to be changed, not a complete color map.

pearsonk, 2018-09-18:
[G512-AllRed-FF0203.zip](https://github.com/MatMoul/g810-led/files/2392883/G512-AllRed-FF0203.zip)
Here is a dump setting all to red (0xFF0203) in one go on G Hub

MatMoul, 2018-09-18:
Thanks for all returns...
I'll try to work on it theses next weeks

MatMoul, 2018-09-24:
@pearsonk I took a look of your dump and it's good dump...
But data is very near that I send...
It's a challenge without the keyboard to test but I'll try...

MatMoul, 2018-12-22:
Thanks @kenthor , the PR #148 correct my big mistake...