MatMoul/g810-led #143: G610 wave effect does not work.

juppso, 2018-09-23:
Enabling any wave effect for the Logitech G610 results in all the keys going black.

Tested using:
sudo g610-led -fx hwave keys 0a

MatMoul, 2018-09-23:
Thanks for the return.

I don't have a g610 and I can't test...
Does this effect work on Windows?
And if yes, is it working when you exit the logitech software?
It's possible that the wave effect was not supported.
And the last question, is the other effects working?

rangelfinal, 2018-10-02:
The effect works as expected on Windows using the official Logitech software, and can be triggered using a special shortcut in the keyboard (Light key + numbers from 1 to 4 trigger different wave types, 5 trigger a special unique effect where waves are created from key presses).  
No effects work on Linux using this project, and all but `cycle all` just turn off all key lights. Cycle all creates a loop where half of the keyboard is turned off and half is turned on, changing every few seconds.  
The G610 is monochrome - maybe the profiles are color based, and G610 expects a different format.

MatMoul, 2019-01-15:
After all the updates from 08/02/2018, have you always this issue ?

duca, 2019-01-23:
I confirm that
`sudo g610-led -fx hwave keys 0a`
does not work on the master branch as expected nor does vwave, waves, breathing. Only color actually works as expected while the rest triggers the "stars" effect.

DanEble, 2019-02-24:
Duca, if you are up for experimenting with your keyboard, consider exploring different values in the message at the end of LedKeyboard::SetNativeEffect().  It looks like this:

	byte_buffer_t data = {
		0x11, 0xff, protocolByte, 0x3c,
		(uint8_t)part, static_cast<uint8_t>(effectGroup),
		// color of static-color and breathing effects
		color.red, color.green, color.blue,
		// period of breathing effect (ms)
		static_cast<uint8_t>(period.count() >> 8), static_cast<uint8_t>(period.count() & 0xff),
		// period of cycle effect (ms)
		static_cast<uint8_t>(period.count() >> 8), static_cast<uint8_t>(period.count() & 0xff),
		static_cast<uint8_t>(static_cast<uint16_t>(effect) & 0xff), // wave variation (e.g. horizontal)
		0x64, // unused?
		// period of wave effect (ms)
		static_cast<uint8_t>(period.count() >> 8), // LSB is shared with cycle effect above
		static_cast<uint8_t>(storage),
		0, // unused?
		0, // unused?
		0, // unused?
	};

Maybe some of those fields that seemed to me to have no effect on my G Pro will have an effect on your G610; or maybe there are even more significant differences in the message format.

DanEble, 2019-02-24:
>EDIT: actually `cycle` also works although within a given effect, the transitions are slow while the effect change is very fast (e.g. hwave propates slowly but the switch between hwave and vwave occurs very fast)

This statement is interesting because you use the name `cycle` but describe the effect `waves`.

duca, 2019-02-24:
> This statement is interesting because you use the name `cycle` but describe the effect `waves`.

Hi DanEble,
Unfortunately I am out town for sometime now and will pass another week before I get to my keyboard again. 

Regarding the description, perhaps my phrasing was unfortunate but what I was trying to achieve is describe how strangely the hwave and vwave effects were behaving when I had cycle enabled. Trying to use them alone actually triggers the star effect as mentioned.

DanEble, 2019-02-24:
> Unfortunately I am out town for sometime now and will pass another week before I get to my keyboard again.

no worries

> Regarding the description, perhaps my phrasing was unfortunate but what I was trying to achieve is describe how strangely the hwave and vwave effects were behaving when I had cycle enabled. Trying to use them alone actually triggers the star effect as mentioned.

I am confused because, as I understand it, cycle, hwave, and vwave are mutually exclusive alternatives.

duca, 2019-02-25:
> 
> I am confused because, as I understand it, cycle, hwave, and vwave are mutually exclusive alternatives.

to my understanding, cycle is not an effect per se, only sequentially activates the actual effects. Perhaps I am mistaken, but when I sent the cycle command, what I saw was vwave for some seconds, then hwave for some seconds and so on.

DanEble, 2019-02-26:
> to my understanding, cycle is not an effect per se, only sequentially activates the actual effects. Perhaps I am mistaken, but when I sent the cycle command, what I saw was vwave for some seconds, then hwave for some seconds and so on.

That is `waves

duca, 2019-05-15:
> Duca, if you are up for experimenting with your keyboard, consider exploring different values in the message at the end of LedKeyboard::SetNativeEffect(). It looks like this:
>
> ```
> byte_buffer_t data [...]

I can do that, but I think I need some guidance here. Perhaps if I manage to get a log from the usb communication between the logitech sw and the keyboard you could assist me on that.

MatMoul, 2020-05-16:
Closing before one year without feedback, you can reopen if you want

closed 2020-05-16 by MatMoul