MatMoul/g810-led #145: GPRO gamemode key color and caps/scrolllock indicators don't work

deathtrip, 2018-10-07:
Basically neither setting an effect for all keys or trying to set them individually work. They are always white.

deathtrip, 2018-10-17:
I tried changing it manually, but with no effect, does it have to be in the config file?
What about using them as part of the effect when they are on?

MatMoul, 2018-10-17:
I use it as this :
`g810-led -fx cycle all 20`
`g810-led -gn indicators 00ff00`
`g810-led -kn game 0000ff`
`g810-led -c`

And in a config file :
`fx cycle all 20`
`g indicators 00ff00`
`k game 0000ff`
`c`

deathtrip, 2018-10-22:
Thanks for the examples!  Is there any way to make these keys follow the effect set for the rest of the keyboard? I tried putting cycle in the config file for them but it didn't work.

MatMoul, 2018-10-22:
No these keys doesn't work with effect...
You can try the natives effects with lightkey + 1 (or 2, 3, 4, ...)...
It's the same result

deathtrip, 2018-10-28:
Now after connecting the keyboard to another machine, back on my main pc, the three keys are always white, i didn't change the config file. Looks like some sort of bug, because everything worked fine before i connected it to the other machine.

deathtrip, 2018-10-31:
Found out it has nothing to do with connecting it to other machines, as now it happened without disconnecting the keyboard. After editing the config file again, it worked for a few days and now the keys are white again. :/