MatMoul/g810-led #151: g610 profile not working

Dherman3607, 2018-12-19:
So I have a g610, and the command via console works for controlling the intensity, however, within the profile file, the same line does not seem to load on reboot, causing me to need to manually input the command each time.

lanodan, 2018-12-19:
I have a Logitech G610 and g610-led version 0.2.7 with libusb and it works here. Are you sure your profile has a `c` line at the end?

The volume control recently broke for me but I think it’s a Xorg driver issue.

Dherman3607, 2018-12-19:
> I have a Logitech G610 and g610-led version 0.2.7 with libusb and it works here. Are you sure your profile has a `c` line at the end?
> 
> The volume control recently broke for me but I think it’s a Xorg driver issue.

here's what my profile looks like:
>#Sample profile by groups keys

>g610-led -a 50
> c # Commit changes

You might be right about the volume mixer.

MatMoul, 2018-12-22:
@Dherman3607 In your profile file, you use :
g610-led -a 50
c
but you need use :
a 50
c

Dherman3607, 2018-12-22:
ok I've made the chance, next time i reboot i'll see if i see a difference.

MatMoul, 2019-01-15:
@Dherman3607 Is this issue can be closed ?