MatMoul/g810-led #153: G Pro has "FN" key in place of right window

DanEble, 2018-12-28:
The UI should use "fn" rather than "win_right" (etc.) to refer to this key.

MatMoul, 2018-12-29:
Thanks for the feedback, I will try to add the fn keyword in addition to win_right

MatMoul, 2019-01-01:
@DanEble As I can see, you work hard...
For the fn you can make a PR for this.
Just edit [the line 167 of helpers/utils.cpp](https://github.com/MatMoul/g810-led/blob/master/src/helpers/utils.cpp#L167)

DanEble, 2019-01-01:
@MatMoul, based on your comment, I infer that you want `fn` to be simply an alias for `win_right`.

What I had in mind when I entered this ticket was that `win_right` should have no effect on a G Pro because there is no key with that function.  It doesn't look to me like utils.cpp is currently set up to enforce such a restriction.

MatMoul, 2019-01-01:
@DanEble I don't know how to set your fn key but for me if you use the win_right, fn is an alias...
if not, fn is an alias of an other key

closed 2019-07-25 by MatMoul