MatMoul/g810-led #157: Set user-stored lighting effect

DanEble, 2018-12-31:
Experimenting with a G Pro, I have discovered how to set the effect that is recalled with "Backlight toggle+7."  In the USB command that sets the native effect, setting the byte at index 16 to 1 stores the effect in the user-stored lighting slot.

I would probably be willing to implement this feature, but I would like guidance on the UI syntax to use.  My instinct is not to add a parameter to the -fx command, but rather to add a new command, say -user-fx.

MatMoul, 2018-12-31:
I'm not sure how to use the backlight+7, on my G810, it seem that has no effect.
Can you best explain what bl+7 make ?

DanEble, 2018-12-31:
The G Pro "user-stored lighting" feature is mentioned at https://support.logitech.com/en_us/product/pro-gaming-keyboard/faq#link_Access-On-Board-lighting-effects-on-the-Pro-Gaming-Keyboard under the heading "Access On-Board lighting effects on the Pro Gaming Keyboard."

After storing an effect, Backlight+7 recalls it.  Also, the same stored effect is used at startup when the startup mode is set to `wave`. (That makes `wave` a poor name in the case of the G Pro, since color, breathing, and cycle effects will also work; but that's a subject for another ticket.)

MatMoul, 2019-01-01:
Ok, but for the name I haven't idea now...
As developer if it's a bit on fx method, -fx is the way
But as user it's not the best way...

DanEble, 2019-01-01:
If you want to keep using `-fx`, then the full command would be something along these lines:

`gpro-led -fx store breathing keys 0000ff 04`
`gpro-led -fx breathing keys 0000ff 04 store`

Brainstorming synonyms: save, persist, commit

MatMoul, 2019-01-01:
Does this function store only effects or also the individually defined keys?

DanEble, 2019-01-01:
This is an option of the command that `LedKeyboard::setNativeEffect()` sends.  When byte 16 of the command is set to 1, then in addition to activating the effect, the keyboard also saves the effect internally so that it can be activated at startup or recalled manually with backlight+7.

I suspect that there is some way to save individually defined keys, but this function is not it.

MatMoul, 2019-01-01:
I did not see this function when I did the reverse engineering ...
Now, regarding the argument, I'm undecided ...
`store` is a good keyword and the two next command are good
`gpro-led -fx store breathing keys 0000ff 04`
`gpro-led -fx breathing keys 0000ff 04 store`
But with this last info, arg can be also
`gpro-led -fx-store breathing keys 0000ff 04`
I leave you the choice of the arg that is the easiest way for the code

MatMoul, 2019-01-02:
Other brainstorming
`g810 -set-bootfx`

DanEble, 2019-01-02:
Am I correct to think that one would not want this option to be applied from a profile, only as a command-line option?

MatMoul, 2019-01-02:
I agree...

closed 2019-01-04 by MatMoul