MatMoul/g810-led #167: How to add repo to ubuntu?

bifunctor, 2019-02-18:
Hi all 
I would like to install g810-led on my ubuntu 18.04. The question is, how to add the repo? 
I tried:
```
sudo apt install g810-led
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package g810-led
```  
Thanks for help.

MatMoul, 2019-02-18:
g810-led is part now of debian and derivative but read the doc :
> Debian (unstable, and 10 or later), Ubuntu 19.04 or later

For other version you need to clone the repo and install it manually

ARitz-Cracker, 2019-04-09:
Yes as @MatMoul said, you do need to install it manually. I've tested this with Ubuntu 14.04 and it's working great. All you have to do is `apt install libusb-1.0-0-dev` and then run `make bin LIB=libusb` while cd'd in the repo. After that, the executable will be in the newly `bin` folder for you to do as you please

MatMoul, 2019-04-09:
@ARitz-Cracker Read "hidapi vs libusb" https://github.com/MatMoul/g810-led/blob/master/INSTALL.md
Try with hidapi first, libusb is very slow

ARitz-Cracker, 2019-04-09:
I did, though libusb was already installed (just not the header files) and hidapi doesn't seem to be as stable on 14.04 (Yes I know I have to upgrade my distro)

ARitz-Cracker, 2019-04-10:
Hey, just thought I should post this, too. It works hidapi works great on 18.04. The package to install is `libhidapi-dev`

MatMoul, 2019-04-10:
@ARitz-Cracker Try `time g810-led` with libusb (more than 300ms) and try it with hidapi (~50ms)
This is why hidapi is the default lib...

ARitz-Cracker, 2019-04-10:
I don't doubt you, dude. I'm all for making things more fast and efficient. That's why I'm reporting that it works flawlessly with Ubuntu 18.04, but not in older versions.

By the way, when I discovered this repo. I was really happy. You've basically destroyed my final excuse not to use Linux as my primary operating system (I really couldn't stand the default colour) I've got a G910 Orion spark and I tried to USB sniff it years ago but I couldn't figure it out at the time and gave up. Though now you've made an application that can interact with it. Not only that, but an open-source library that can be used to create any effects I'd like. So for that, I really thank you.

norbet-sommud, 2019-07-04:
I'm to the point of having a folder structure of the cloned git library on my desktop and I don't know what to do from there.  If you guys could give me some terminology to google for what I'm trying to do, I'm sure I can figure it out from there.  I just don't know how to describe "installing from a git clone" well enough to get anything useful at present.

ARitz-Cracker, 2019-07-04:
> I'm to the point of having a folder structure of the cloned git library on my desktop and I don't know what to do from there. If you guys could give me some terminology to google for what I'm trying [...]

What you're trying to do is "compile from source"

Step 1: Open a terminal to your favorite folder, run `git clone https://github.com/MatMoul/g810-led.git`
Step 2: `cd g810-led` 
Step 3: Assuming you're on a Ubuntu-based Distro, install `libhidapi-dev`, and `build-essential`
Step 4: As stated in the [instructions](https://github.com/MatMoul/g810-led/blob/master/INSTALL.md) run `make bin`
Step 5: The compiled program is now in the newly created `bin` folder. You can also run `sudo make install` while in the `g810-led` folder to globally install this program in your system if that's what you want.

However, some packages are already in their respective package managers. Check out the previously mentioned install instructions to learn more.

Added note: Most of the time when you're compiling software from github, the process is usually the same:
1. Clone the repo and open a terminal in its folder
2. Run `./configure` if it exists.
3. Run `make` unless otherwise specified in the individual repos instructions
4. Run `make install` to globally install the software to the system.
Sidenote: the `make` program reads the `makefile` file in the current directory. That's how it knows what software to work with

You can also run `sudo make uninstall` while in the project directory to uninstall the program.

ARitz-Cracker, 2019-07-04:
Oh and I almost forgot, for larger projects with lots of files to compile, a good tip is to add `-j 16` to the end of the first `make` command. e.g. `make -j 16`. Replace `16` with how many threads your CPU has. This will compile the software in multi-threaded mode. Making it to faster.

norbet-sommud, 2019-07-04:
Thank you.  I guess I had actually done it - I just had not tried to execute the program via the terminal.  I'm an idiot.

I'll work on getting to the level where the CPU threading has bearing on my output.  Much appreciated.

MatMoul, 2019-09-23:
Seem solved

closed 2019-09-23 by MatMoul