MatMoul/g810-led #169: G513 Silver Not Working

ericscottmarquez, 2019-03-13:
G513 Silver , NOT carbon
installed through the repo with make 
Tried the lsusb thing:
```
Bus 001 Device 010: ID 046d:c33c Logitech, Inc. 
g513-led -dv 046d -dp c33c -tuk 1 -a 000000 <- matching or compatible device not found!
g513-led -dv 046d -dp c33c -tuk 2 -a 000000 <- this ones does not return an error, but lights don't change
g513-led -dv 046d -dp c33c -tuk 3 -a 000000 <- same thing, this ones does not return an error, but lights don't change

g513-led -a 39ff14 <- tried with all three protocols, randomly flashes
green then immediately changes back. Not even the glitch is consistent.
```
When I installed g810-led, the keyboard took a default layout color-scheme but that's it.
None of the commands work.

How can I get this working?

MatMoul, 2019-03-15:
If the default color-scheme work, your keyboard is supported....
It seem you don't have right to run g810-led as user, try with sudo or as root...
If you want to use it as user, add the udev rules.

ericscottmarquez, 2019-03-15:
Sorry, I've been stressed with other projects and gave a half-baked post.

What happens is that I tested all commands as root / sudo and the only one that responds is the -a color command, but it's only responding as a glitch. Withj color-change commands, it only flashes for about 100ms , barely enough to notice, and then it returns to the default configuration. It Seems like there is a glitch with this keyboard. How can I fix it or contribute to development?

using ubuntu 18.04 with KDE plasma.

MatMoul, 2019-03-16:
Is your keyboard connected to an USB hub ?
It has an issue for this #114

msagerup, 2019-04-12:
I had the same issue on the G513

Menthelice, 2019-06-06:
I have the same issue with G413 Carbon. Also like Eric says the -a color command works.

norbet-sommud, 2019-07-04:
Same issue - also through a KVM switch/hub - sudo resolved.  Thank you again for developing this.

MatMoul, 2019-07-05:
@msagerup  @norbet-sommud Thanks for your returns, it would be nice if @ericscottmarquez gave us a little feedback...

If you want to skip the sudo, take a look at the udev rules...
https://github.com/MatMoul/g810-led/blob/master/udev/g810-led.rules

MatMoul, 2019-09-23:
Closing because no feedback
You can reopen it or open a new issue

closed 2019-09-23 by MatMoul