MatMoul/g810-led #174: Support for osx

martinjuhasz, 2019-06-05:
Would love to see support for osx on this.
I know the official app supports osx but it just works half of the time.

currently getting
`clang: error: cannot specify -o when generating multiple output files`
when trying to build from source

MatMoul, 2019-06-06:
I've just tried a search with your error and I found this :
https://stackoverflow.com/questions/46527662/xcode-9-clang-error-cannot-specify-o-when-generating-multiple-output-files

I don't have Mac to test it...

martinjuhasz, 2019-06-10:
Sadly not. This is for building a project with xcode. I tried different solutions that i found online without luck.

martinjuhasz, 2019-06-10:
i was able to build it, still i'm getting a `Matching or compatible device not found!` error.

First install the dependencies ofc:
```
brew install lhidapi
# or
brew install libusb
```

and then changing the following lines in `makefile`
```
-	LIBS=-lhidapi-hidraw
+	LIBS=-lhidapi

-APPSRCS=src/main.cpp src/helpers/*.cpp src/helpers/*.h
-LIBSRCS=src/classes/*.cpp src/classes/*.h
+APPSRCS=src/main.cpp src/helpers/*.cpp
+LIBSRCS=src/classes/*.cpp
```
because of following [this](https://stackoverflow.com/questions/43525486/cannot-specify-o-when-generating-multiple-output-files-c-error) SO answer.

I think we can close this issue for the moment. We would need an active developer on this tool with an osx machine to trace down the problem. Thanks for this library anyway!

MatMoul, 2019-06-11:
if you was able to build it, try :
`g810-led --list-keyboards`
if you receive the same message, it's don't work but otherwith, this is because you need to use it as root with sudo.

martinjuhasz, 2019-06-11:
still no luck, but it lists my keyboard!
```
$ ./g910-led --list-keyboards
Device: Logitech - Gaming Keyboard G910
	Vendor ID: 046d
	Product ID: c335
	Serial Number: 107F37583233
Device: Logitech - Gaming Keyboard G910
	Vendor ID: 046d
	Product ID: c335
	Serial Number: 107F37583233
$ sudo ./g910-led -a 00ff00
Matching or compatible device not found !
```

MatMoul, 2019-06-12:
@martinjuhasz Sorry to hear that
I tag tp `help wanted`

martinjuhasz, 2019-06-13:
i just saw people having problems with usb-c type connections. i'm also running this through an adapter to a usb-c input, so probably thats the current issue i'm facing and not related to osx anymore since i've built it successfully.

MatMoul, 2019-06-20:
@martinjuhasz you're right, it's a possibility....

MatMoul, 2019-10-16:
New PR for OSX : #193

arpitest, 2020-01-04:
I've got it working on OSX Mojave 10.14.6:
makefile did not work at all, but compiling this way:
$ cd src
$ c++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.3.9\" main.cpp */*.cpp -lhidapi

sometimes it is unable to set the color but repeating the command a few times fixes it. maybe some built-in retry option could help, but i used this workaround for G213:

while ! sudo ./a.out -r 1 ffffff ; do sleep 1; done

Digicrat, 2020-03-16:
I got it to compile, building on the information above, but I'm also not seeing any devices when I run "g810-led --list-keyboards".  This is a G915 keyboard. I tested with all 3 interface options (BT, Lightspeed, and wired).  And now I see that #198 likely covers at least part of that problem.

Compilation Pre-requisite:  "brew install hidapi"

I posted my changes to a PR - #218

naprsa, 2021-10-01:
Hi! I installed hidapi with brew (osx 11.6), but still can't to build app. Here is log. Can anybody help me?

```
➜  g810-led git:(master) ✗ sudo make bin
c++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.4.2\"  src/main.cpp src/helpers/help.cpp src/helpers/utils.cpp src/classes/Keyboard.cpp -o bin/g810-led -lhidapi
In file included from src/main.cpp:24:
In file included from src/helpers/utils.h:22:
src/helpers/../classes/Keyboard.h:162:17: warning: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Wnon-c-typedef-for-linkage]
                typedef struct {
                              ^
                               DeviceInfo
src/helpers/../classes/Keyboard.h:163:4: note: type is not C-compatible due to this default member initializer
                        uint16_t vendorID = 0x0;
                        ^~~~~~~~~~~~~~~~~
src/helpers/../classes/Keyboard.h:169:5: note: type is given name 'DeviceInfo' for linkage purposes by this typedef declaration
                } DeviceInfo;
                  ^
1 warning generated.
In file included from src/helpers/help.cpp:20:
In file included from src/helpers/utils.h:22:
src/helpers/../classes/Keyboard.h:162:17: warning: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Wnon-c-typedef-for-linkage]
                typedef struct {
                              ^
                               DeviceInfo
src/helpers/../classes/Keyboard.h:163:4: note: type is not C-compatible due to this default member initializer
                        uint16_t vendorID = 0x0;
                        ^~~~~~~~~~~~~~~~~
src/helpers/../classes/Keyboard.h:169:5: note: type is given name 'DeviceInfo' for linkage purposes by this typedef declaration
                } DeviceInfo;
                  ^
1 warning generated.
In file included from src/helpers/utils.cpp:17:
In file included from src/helpers/utils.h:22:
src/helpers/../classes/Keyboard.h:162:17: warning: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Wnon-c-typedef-for-linkage]
                typedef struct {
                              ^
                               DeviceInfo
src/helpers/../classes/Keyboard.h:163:4: note: type is not C-compatible due to this default member initializer
                        uint16_t vendorID = 0x0;
                        ^~~~~~~~~~~~~~~~~
src/helpers/../classes/Keyboard.h:169:5: note: type is given name 'DeviceInfo' for linkage purposes by this typedef declaration
                } DeviceInfo;
                  ^
1 warning generated.
In file included from src/classes/Keyboard.cpp:17:
src/classes/Keyboard.h:162:17: warning: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Wnon-c-typedef-for-linkage]
                typedef struct {
                              ^
                               DeviceInfo
src/classes/Keyboard.h:163:4: note: type is not C-compatible due to this default member initializer
                        uint16_t vendorID = 0x0;
                        ^~~~~~~~~~~~~~~~~
src/classes/Keyboard.h:169:5: note: type is given name 'DeviceInfo' for linkage purposes by this typedef declaration
                } DeviceInfo;
                  ^
1 warning generated.
ld: library not found for -lhidapi
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [bin/g810-led] Error 1
```
```
➜  g810-led git:(master) ✗ brew install hidapi
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
Updated 8 formulae.

Warning: hidapi 0.10.1 is already installed and up-to-date.
To reinstall 0.10.1, run:
  brew reinstall hidapi
```