MatMoul/g810-led #178: bug with G213: Matching or compatible device not found !

Strubbl, 2019-07-23:
i am using this software [g810-led-git](https://aur.archlinux.org/packages/g810-led-git/) in Archlinux. Since the latest release i activate the color on my keyboard anymore. Before that it worked very well. Please see my attempt to set the color:

```
strubbl@carrie:~$ g213-led -a 449aff
Matching or compatible device not found !
strubbl@carrie:~$ g213-led --list-keyboards
Device: Logitech - Gaming Keyboard G213
	Vendor ID: 046d
	Product ID: c336
	Serial Number: 0269367A3132
strubbl@carrie:~$ g213-led --print-device
Matching or compatible device not found !

```

I also started the command with strace if that helps: https://gist.github.com/Strubbl/d59780fb4d6e7383be271a2497df56a0

Strubbl, 2019-07-23:
Because i saw in strace a permissions problem, i just started this command with sudo and it works.

https://gist.github.com/Strubbl/d59780fb4d6e7383be271a2497df56a0#file-gistfile1-txt-L2046

So is this a bug?

Strubbl, 2019-07-23:
Is it possible to evaluate the error code and give a better error message to the user?
Is there a special user group, which my user user should be part of to avoid the permission problem?

MatMoul, 2019-07-23:
@Strubbl The last release include changes in udel rules...
The PR : https://github.com/MatMoul/g810-led/pull/176/files
Feedbacks are welcome...

davidgasquez, 2019-07-24:
Hey there! Same happening to me with a G610. 
```
$ g610-led --list-keyboards
Device: Logitech - Gaming Keyboard G610
	Vendor ID: 046d
	Product ID: c333
	Serial Number: 0B8A354F3430
$ g610-led  --print-device
Matching or compatible device not found !
```

Let me know if I can send something else like a debug log or anything!

graves501, 2019-07-24:
Same issue here with Logitech G Pro:
```
$ gpro-led --list-keyboards
Device: Logitech - Pro Gaming Keyboard
        Vendor ID: 046d
        Product ID: c339
        Serial Number: 0782366A3537
$ gpro-led --print-device
Matching or compatible device not found !
```

Used to work 2-3 weeks ago, so something probably went wrong with the recent version?

Let me know, if you need logs or other output.

Thank you!

MatMoul, 2019-07-24:
As I say in my previous post, the last change look at this PR : https://github.com/MatMoul/g810-led/pull/176/files
The main changes is that we have replaced SUBSTEMS by SUBTEM
Feedbacks are always welcome...

Strubbl, 2019-07-24:
I am not familiar with udev. Feedback is, that the change might cause the problem.
What is the difference between SUBSTEMS and SUBSTEM?

MatMoul, 2019-07-24:
@Strubbl The difference between SUBSTEMS and SUBSTEM was discussed in the PR, your welcome if you encouter issues...
Sorry for my last link, look at here : https://github.com/MatMoul/g810-led/pull/176

graves501, 2019-07-25:
So I forked your repo and reverted the PR (commit `3c06d8f952e1617affe4099c068943a92baf8bf7`). 
Now everything works fine again, so it has something to do with the SUBSYSTEM.

My system runs `Manjaro 18.0.4 Illyria` with Kernel `4.19.59-1`.

MatMoul, 2019-07-25:
Thanks for all your feedbacks, new release here and should fixe this issue...

closed 2019-07-25 by MatMoul