MatMoul/g810-led #179: Issue after customize

hoangnguyen7699, 2019-07-24:
Hi! After I customize my lighting keyboard, I type "sudo gpro-led -c" to commit all changes but when I restart my computer, it does not change anything. Do you have solution for me?
Thanks

MatMoul, 2019-07-24:
Hi, g810-led only set the profile...
But Logitech G keyboard dont have memory, you need to set the profile eatch time you boot...
This is why you can set your profile in /etc/g810-led/profile...
With the udev rules, it set you profile when you boot or you connect your keybaord.

hoangnguyen7699, 2019-07-24:
can you say more detail what step I need to do? Because my major is not IT, so I could not understand!
Thanks

MatMoul, 2019-07-24:
@hoangnguyen7699 It's not so easy to help you with these infos...
What's your distro and how do you install g810-led?
The project provide an udev rules file.
This file run the g810-led bin when you start your computer or connect your keyboard...

hoangnguyen7699, 2019-07-24:
My situation is: I have logitech gpro keyboard, I use your instruction to custom light but when I restart my computer, it always show default light. How I can change light and after I turn on my computer, it always use it as default?
Sorry if I bother you! Thanks

MatMoul, 2019-07-24:
@hoangnguyen7699 Try to look at /etc/g810-led/profile, this is the file that set your keyboard.
To launch this file you need a udev rule file. Whis this, your keyboard as set when you boot or you connect it...
I ask you a second time : What's your distro and how do you install g810-led?

hoangnguyen7699, 2019-07-24:
I used these commad to install:
sudo apt-get install git g++ make libhidapi-dev
git clone https://github.com/MatMoul/g810-led.git
cd g810-led
make bin
sudo make install

After that, I type: gpro-led --help
Then I followed your instruction there to custom.
That 's all what I did.
Thanks very much!

hoangnguyen7699, 2019-07-24:
my OS is ubuntu 18.04 and my keyboard is logitech Gpro

MatMoul, 2019-07-24:
@hoangnguyen7699 I don't know very well ubuntu releases...
But with Debian10 you can install g810-led with `apt install g810-led`
If this doesn't work for you, you need to look at the `/etc/udev/rules.d/g810-led.rules`

hoangnguyen7699, 2019-07-24:
Let me try! Thanks a lot

hoangnguyen7699, 2019-07-25:
My OS version is 18.04, so I could not install in that way. File `/etc/udev/rules.d/g810-led.rules` looks like this: 
[image]

So can you show teach me how I can fix my problem?
Sorry if I bother you!

MatMoul, 2019-07-25:
Have you edited /etc/g810-led/profile?

hoangnguyen7699, 2019-07-25:
I dont have that file nmed 'etc'. What shoud I do?

MatMoul, 2019-07-25:
If you're on linux, you have an /etc directory, this is the config directory...
Try this:
`cd /etc/g810-led`
`sudo nano profile`

hoangnguyen7699, 2019-07-25:
Yes, I got it! Thanks very much! You're nice:))

closed 2019-07-25 by MatMoul