MatMoul/g810-led #182: udev rules with `SUBSYSTEMS=="usb"` breaks Multimedia keys

lanodan, 2019-08-29:
With https://github.com/MatMoul/g810-led/commit/a6586266f1c1ffdec4d934a8d6b798dc70fd7697 my multimedia keys (mute, volume, play/pause, stop, prev, next) on my G610 keyboard aren't working at all (no events with `evtest`, `xev` or wayland).

They got fixed with <https://github.com/MatMoul/g810-led/commit/3c06d8f952e1617affe4099c068943a92baf8bf7> according to the commit message it fixes repeats messages in dmesg, which I only get when g610-led command fails (like when `/etc/g810-led/profile` is missing).

Also I can confirm that the udev rules are still being run for me without the `SUBSYSTEMS=="usb"` lines.

MatMoul, 2019-08-29:
@lanodan I'm not sure what your issue but g810-led don't handle multimedia keys action, only led...
For udev rules, it has two functions :
- Set the profile when you connect your keyboard or start your linux
- Use g810-led as user
About SUBSYSTEM/SUBSTEMS, some distro need the first and other the second

lanodan, 2019-08-30:
[2019-08-29 13:22:24-0700] MatMoul:
> @lanodan I'm not sure what your issue but g810-led don't handle multimedia keys action, only led...

Yes, I don't think it's because of g810-led (the program) but more like 
the udev rule alone which might be overriding/crashing on other stuff 
like evdev/libinput stuff used for detecting the keyboard additionnal 
keys.

> For udev rules, it has two functions :
> - Set the profile when you connect your keyboard or start your linux
> - Use g810-led as user
> About SUBSYSTEM/SUBSTEMS, some distro need the first and other the second

I guess I'll look into how it should work for Gentoo.

MatMoul, 2019-08-30:
@lanodan For info, I use ArchLinux with libinput but a package is available too in Debian 10 and child distros.
If you don't use udev, it's sure you need an other solution than udev

NerdHarder615, 2019-09-11:
I had a similar issue with this today. I was able to resolve the issue by installing this package:
https://aur.archlinux.org/packages/sidewinderd/

Not sure if it will fix your issues, but it is worth a shot. FYI I am using Manjaro

lanodan, 2019-09-11:
[2019-09-11 06:00:28-0700] Matt:
> I had a similar issue with this today. I was able to resolve the issue by installing this package:
> https://aur.archlinux.org/packages/sidewinderd/
> 
> Not sure if it will fix your issues, but it is worth a shot. FYI I am using Manjaro

I don't think so, I have a G610 (orion brown) and this one doesn't seems 
to have anything udev related (it looks like a daemon kind of thing…)

---

I checked the udev documentation and tried a bunch of stuff, I'm not 
really sure what's going on and AFAIK the behaviour between udev and eudev 
should be the same, and neither ArchLinux or Gentoo are keen on patching 
stuff other than doing fixes.
I guess I'll check with adding ArchLinux on my machine and a 
systemd-udev on a gentoo.

MatMoul, 2019-09-23:
@lanodan Is this issue always relevant?
I've see that a package is available for Gentoo :
https://repology.org/project/g810-led/versions
https://packages.gentoo.org/packages/app-misc/g810-led

lanodan, 2019-09-23:
[2019-09-23 13:05:24-0700] MatMoul:
> @lanodan Is this issue always relevant?

Yeah, it's still relevant, haven't got the time to install an ArchLinux on one of my machines for it yet.

> I've see that a package is available for Gentoo :
> https://repology.org/project/g810-led/versions
> https://packages.gentoo.org/packages/app-misc/g810-led

Yup, I maintain it.

MatMoul, 2019-09-23:
@lanodan Thanks to maintain it, issue will still open...

MatMoul, 2020-04-26:
No return, I close but you can reopen if you want

closed 2020-05-16 by MatMoul