MatMoul/g810-led #183: color change does not work

nonthevisor, 2019-09-04:
**System**
```
$ uname -a
Linux port-5325 4.9.0-9-amd64 #1 SMP Debian 4.9.168-1+deb9u5 (2019-08-11) x86_64 GNU/Linux
$ pwd
/usr/local/bin
```
**Problem**
I cannot change the colors of my keys, thus leaving me with the rather distracting rainbow effect. 

**Background**
First, I tried to install by package
```
$ apt search g810-led
Sorting... Done
Full Text Search... Done
```
However, I couldn't find it, so I tried to install it via git:
```
git clone https://github.com/MatMoul/g810-led.git
cd g810-led
make bin # for hidapi
```
This gave me the following error
`Error: Can not write to hidraw, try with the libusb version`
This surprised me, because I have libhidapi installed:
```
$ apt show libhidapi-dev
Package: libhidapi-dev
Version: 0.8.0~rc1+git20140818.d17db57+dfsg-1
Priority: extra
Section: libdevel
Source: hidapi
Maintainer: Scott Talbert <swt@techie.net>
Installed-Size: 193 kB
Depends: libhidapi-libusb0 (= 0.8.0~rc1+git20140818.d17db57+dfsg-1), libhidapi-hidraw0 (= 0.8.0~rc1+git20140818.d17db57+dfsg-1)
Homepage: http://www.signal11.us/oss/hidapi/
Tag: devel::library, role::devel-lib
Download-Size: 45.8 kB
APT-Manual-Installed: yes
APT-Sources: http://ftp.de.debian.org/debian stretch/main amd64 Packages
Description: Multi-Platform library for communication with HID devices (development files)
 HIDAPI is a multi-platform library enabling applications to easily
 interface with Bluetooth and USB HID-class devices.
 .
 This package contains the files needed to compile and link programs which
 use HIDAPI.
```
But I did not feel like getting to the core of this, so instead I deleted `bin` and ran 
```
$ sudo make bin LIB=libusb
g++ -Dlibusb -Wall -O2 -std=gnu++11 -DVERSION=\"0.3.8\"  src/main.cpp src/helpers/utils.cpp src/helpers/help.cpp src/classes/Keyboard.cpp -o bin/g810-led -lusb-1.0
```
So far, so good. However, installing failed:
```
$ sudo make install
makefile:85: recipe for target 'install' failed
make: *** [install] Error 1
```

Now, I thought that while convenient, I don't really need to install, because I can use it via `./g810-led`, which also worked to some degree:
```
sudo ./g810-led --list-keyboards
Device: Logitech - Gaming Keyboard G910
	Vendor ID: XXXX
	Product ID: XXXX
	Serial Number: XXXXXXXXXXXX
```
**Main problem**
But it does not work when trying to change the colors:
`$ sudo ./g810-led -a 00ccff`
No error message, just a lot of newlines and no change in color. 

**Related issues**
[sudo make install fails](https://github.com/MatMoul/g810-led/issues/136)
Sadly, running uninstall and then install did not work for me.
[How to use this file. Can someone give an example?](https://github.com/MatMoul/g810-led/issues/140)
I thought this might still work without a successful install, but it did not.

MatMoul, 2019-09-04:
g810-led use now hidapi but on CentOS, I've receive the same message `Error: Can not write to hidraw, try with the libusb version`
This why you can use libusb but you need to install libusb-dev before making g810-led...
I've tryied to find why I can't write to hidraw on CentOS but without sucsess...
selinux, apparmor or other, that is the question ?

nonthevisor, 2019-09-05:
I tried again, but couldn't solve the issue.
First, I installed libusb-dev
`$ sudo apt install libusb-dev`
So by now I have the following libusb packages installed:
```
libgusb2/oldstable,now 0.2.9-1+b1 amd64 [installed,automatic]
  GLib wrapper around libusb1

libhidapi-libusb0/oldstable,now 0.8.0~rc1+git20140818.d17db57+dfsg-1 amd64 [installed,automatic]
  Multi-Platform library for communication with HID devices (libusb backend)

libhpmud0/oldstable,now 3.16.11+repack0-3 amd64 [installed,automatic]
  HP Multi-Point Transport Driver (hpmud) run-time libraries

libusb-0.1-4/oldstable,now 2:0.1.12-30 amd64 [installed,automatic]
  userspace USB programming library

libusb-1.0-0/oldstable,now 2:1.0.21-1 amd64 [installed,automatic]
  userspace USB programming library

libusb-1.0-0-dev/oldstable,now 2:1.0.21-1 amd64 [installed]
  userspace USB programming library development files

libusb-1.0-doc/oldstable,now 2:1.0.21-1 all [installed,automatic]
  documentation for userspace USB programming

libusb-dev/oldstable,now 2:0.1.12-30 amd64 [installed]
  userspace USB programming library development files

libusbmuxd4/oldstable,now 1.0.10-3+b1 amd64 [installed,automatic]
  USB multiplexor daemon for iPhone and iPod Touch devices - library

libusbredirhost1/oldstable,now 0.7.1-1 amd64 [installed,automatic]
  Implementing the usb-host (*) side of a usbredir connection (runtime)

libusbredirparser1/oldstable,now 0.7.1-1 amd64 [installed,automatic]
  Parser for the usbredir protocol (runtime)

```
Second, I deleted the whole directory to start from scratch, then making with the libusb option set, and then make install:
```
$ sudo git clone https://github.com/MatMoul/g810-led.git
Cloning into 'g810-led'...
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 1620 (delta 2), reused 7 (delta 2), pack-reused 1612
Receiving objects: 100% (1620/1620), 863.46 KiB | 0 bytes/s, done.
Resolving deltas: 100% (787/787), done.
$ cd g810-led/
$ sudo make bin LIB=libusb
g++ -Dlibusb -Wall -O2 -std=gnu++11 -DVERSION=\"0.3.8\"  src/main.cpp src/helpers/utils.cpp src/helpers/help.cpp src/classes/Keyboard.cpp -o bin/g810-led -lusb-1.0
$ sudo make install
makefile:85: recipe for target 'install' failed
make: *** [install] Error 1
```

Still, the same error. 
Trying to use it without making install failed in the same way it did before, meaning I could list the keyboard but not change the color, without any error message occurring.

MatMoul, 2019-09-07:
About your distribution `Debian 9`, I remember having tested without problem g810-led with hidapi.
I'm pretty sure you have a package or configuration that has the same effect as CentOS.
Could you tell me more about your installation?

I'll try to test on the same distro when I'll be on my desktop...

nonthevisor, 2019-09-10:
Could you tell me what information you need? Besides listing my libusb and hidapi packages, I'm not really sure what could be relevant information. Since I already posted my libusb packages, here are my hidapi packages:
```
libhidapi-dev/oldstable,now 0.8.0~rc1+git20140818.d17db57+dfsg-1 amd64 [installed]
  Multi-Platform library for communication with HID devices (development files)
libhidapi-hidraw0/oldstable,now 0.8.0~rc1+git20140818.d17db57+dfsg-1 amd64 [installed,automatic]
  Multi-Platform library for communication with HID devices (hidraw backend)
libhidapi-libusb0/oldstable,now 0.8.0~rc1+git20140818.d17db57+dfsg-1 amd64 [installed,automatic]
  Multi-Platform library for communication with HID devices (libusb backend)
```

MatMoul, 2019-09-10:
uname -a is very strange...
What's "Linux port-5325"?
What's your distro and DE?
I've just started my debian but the uname isn't the same "Linux debian"

nonthevisor, 2019-09-11:
"port-5325" is just my network hostname. your network hostname is probably "debian", which is quite common for hosts that are not part of a company network or similar and therefore don't need unique names. 
the output of `uname -a` is:
 `kernel-name; network hostname; kernel-release; kernel-version; machine hardware name; operating system` 
in my case that's:
 `Linux; port-5325; 4.9.0-11-amd64; #1 SMP Debian 4.9.189-3 (2019-09-02); x86_64; GNU/Linux`

see the [man page](https://linux.die.net/man/1/uname) for more details

As to DE and distro:
```
$ cat /etc/*-release
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

$ echo $XDG_CURRENT_DESKTOP
GNOME
```

MatMoul, 2019-09-11:
@nonthevisor yeah it's the hostname, the shame :(

On my debian 9 with a minimal install, I can use hidraw...

I'll try soon with a full debian gnome if I can reproduce your first error :
`Error: Can not write to hidraw, try with the libusb version`

MatMoul, 2019-09-14:
@nonthevisor I've just tested with a clean install with the graphic installer of Debian 9 and checked gnome.
After that, I've made these follow commands :
`sudo apt-get install git g++ make libhidapi-dev`
`wget https://github.com/MatMoul/g810-led/archive/v0.3.8.tar.gz`
`tar -xf v0.3.8.tar.gz`
`cd g810-led-0.3.8`
`make bin`
`sudo make install`
And it work...

Try to take a look on #114 
Or check if you use a security solution on your Debian

nonthevisor, 2019-09-16:
The [issue](https://github.com/MatMoul/g810-led/issues/114) you linked helped, since my keyboard is connected via a KVM switch. Connecting it directly stopped the rainbow effect. 

I deleted all the files from my previous attempts and installed it again the way you proposed in your most recent [answer](https://github.com/MatMoul/g810-led/issues/183#issuecomment-531495717) (without reinstalling the listed packages, since they were already up to date), and the install went through successfully, this time. 

Now, when connected via the **KVM switch**, I cannot change the colors, but the rainbow effect stopped, so I just have a colorful keyboard. Trying to change the colors results in `Error: Can not write to hidraw, try with the libusb version`. So it seems that somehow using the KVM switch messes with package compatibility, although I cannot imagine why.

When connected **directly**, I can change the colors, with the exception of three keys. This is probably a different issue though, so you can probably close this one.

Thank you very much for you support, and the work you put into maintaining the project.

pearsonk, 2019-09-16:
@nonthevisor Most KVMs don't simply switch/relay connected keyboards and mice. What they do is an intercept-emulate model where they keep a constant generic keyboard and mouse interface to each connected system. Unfortunately, this means that they block any special capabilities or other interfaces for connected keyboards and mice. The reason they do this is for performance reasons, since constant hotplug events that would be otherwise required can cause significant delays. It is easier to simply have a constant device present to each computer, and simply multiplex the input to the selected computer. There are KVMs that will simply switch connected USB devices, but those are very expensive.

nonthevisor, 2019-09-16:
@pearsonk thank you, I never thought about how the switch works. It makes sense then, that it couldn't just work like it would for a direct connection.

MatMoul, 2019-09-20:
@nonthevisor Thanks for the feedback, when I asked 
> Could you tell me more about your installation?
I had suspected a specialty...
Personally, I use KVM consoles in my work but with time, they age very bad despite the price :(

closed 2019-09-20 by MatMoul