MatMoul/g810-led #185: G510s support

Lyberta, 2019-09-08:
`046d:c22d Logitech, Inc. G510 Gaming Keyboard`

This keyboard can only do one color for all the keys at the same time. Is there a tutorial so I can provide dumps?

MatMoul, 2019-09-08:
Before making dump, do you have tested existing protocols :
https://github.com/MatMoul/g810-led#testing-unsupported-keyboards-

Lyberta, 2019-09-08:
So far it only turned off caps lock, num lock and scroll lock indicators. I had to restart the keyboard to get them working again.

MatMoul, 2019-09-08:
After some research, it seem that G510s is an old keyboard with the previous controller...
On the ArchWIKI https://wiki.archlinux.org/index.php/Logitech_Gaming_Keyboards G510S is partialy supported by Gnome15.
An other idea : https://github.com/search?q=g510

pearsonk, 2019-09-09:
I've done some work on this device before. Of note:
- There are 2 USB IDs, depending on whether or not Audio is connected (046d:c44d and 046d:c22e)
- The same USB IDs work for the G510 and G510s models

The raw data for LED control is very simple (0x05RRGGBB). I didn't explore any effects or other possible functions to fuzz the device, since I only cared about the core RGB setting. I know the LCD canvas handling is a bitmap flush, but is out of scope for this project.

The reason I didn't go through the effort of integrating into this project was that it would make code a bit messy for a very old device that is no longer available for purchase.

I have used libg15 and libg15render in the past. They work fine for anything that is a "call & quit" type of program (if you handle the kernel driver reattachment yourself), but I do not recommend integrating into any type of system daemon. Also, it is no longer maintained so any problems you encounter will not be fixed.

MatMoul, 2019-09-09:
@pearsonk Thanks for your always valuable comments...

@Lyberta It seem not very well to include this keyboard on this project...
If I'll add other Keyboard series, I'll start a new project or a full refactoring of the code, more versatile, that will support other brands too but not this year...

Lyberta, 2019-09-09:
@pearsonk Any specific repos you can recommend? I want to try to build whatever code is still there and try it.

pearsonk, 2019-09-09:
@Lyberta https://github.com/Mike-Petersen/g15tools is the root I tried. It has a few custom fixes applied. There are still some resource leaks over time and can cause problems when using alongside anything else that tries to use syslog. If you are using syslog, make sure to initialize the libg15 prior to setting up your applications logging or it will crash.

closed 2019-09-23 by MatMoul

Lyberta, 2019-11-19:
FYI there is now a [proposed kernel driver](https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h=for-5.5/logitech) for linux 5.5.

I've built a custom kernel with it and it works. Provides files in `/sys` to change the color and brightness.

Here's the udev rules file I've made so it can be changed by a user in `plugdev` group:
```
KERNEL=="g15::kbd_backlight", SUBSYSTEM=="leds", ACTION=="add", RUN+="/usr/bin/find /sys$devpath -type f -exec /usr/bin/chgrp plugdev {} ; -exec /usr/bin/chmod 0660 {} ;"
KERNEL=="g15::power_on_backlight_val", SUBSYSTEM=="leds", ACTION=="add", RUN+="/usr/bin/find /sys$devpath -type f -exec /usr/bin/chgrp plugdev {} ; -exec /usr/bin/chmod 0660 {} ;"

KERNEL=="g15::macro_preset1", SUBSYSTEM=="leds", ACTION=="add", RUN+="/usr/bin/find /sys$devpath -type f -exec /usr/bin/chgrp plugdev {} ; -exec /usr/bin/chmod 0660 {} ;"
KERNEL=="g15::macro_preset2", SUBSYSTEM=="leds", ACTION=="add", RUN+="/usr/bin/find /sys$devpath -type f -exec /usr/bin/chgrp plugdev {} ; -exec /usr/bin/chmod 0660 {} ;"
KERNEL=="g15::macro_preset3", SUBSYSTEM=="leds", ACTION=="add", RUN+="/usr/bin/find /sys$devpath -type f -exec /usr/bin/chgrp plugdev {} ; -exec /usr/bin/chmod 0660 {} ;"
KERNEL=="g15::macro_record", SUBSYSTEM=="leds", ACTION=="add", RUN+="/usr/bin/find /sys$devpath -type f -exec /usr/bin/chgrp plugdev {} ; -exec /usr/bin/chmod 0660 {} ;"
```

And some C++ to change it:
`Backlight.h`:
```
#pragma once

#include <fstream>

namespace G510s
{

struct Color
{
	std::uint8_t red;
	std::uint8_t green;
	std::uint8_t blue;
};

class Backlight final
{
public:
	Backlight();
	Color GetDefaultColor();
	void SetDefaultColor(Color color);
	Color GetColor();
	void SetColor(Color color);
private:
	std::fstream m_default_color_file;
	std::fstream m_color_file;

	static Color GetColorInternal(std::fstream& stream);
	static void SetColorInternal(std::fstream& stream, Color color);
};

}
```

`Backlight.cpp`:
```
#include "Backlight.h"

#include <sstream>
#include <iomanip>

namespace G510s
{

Backlight::Backlight()
	: m_default_color_file{"/sys/class/leds/g15::power_on_backlight_val/color"},
	m_color_file{"/sys/class/leds/g15::kbd_backlight/color"}
{
	m_default_color_file.exceptions(
		std::ios_base::failbit | std::ios_base::badbit);
	m_color_file.exceptions(std::ios_base::failbit | std::ios_base::badbit);
}

Color Backlight::GetDefaultColor()
{
	return GetColorInternal(m_default_color_file);
}

void Backlight::SetDefaultColor(Color color)
{
	SetColorInternal(m_default_color_file, color);
}

Color Backlight::GetColor()
{
	return GetColorInternal(m_color_file);
}

void Backlight::SetColor(Color color)
{
	SetColorInternal(m_color_file, color);
}

Color Backlight::GetColorInternal(std::fstream& stream)
{
	std::string color_string;
	stream >> color_string;
	auto r = static_cast<std::uint8_t>(std::stoi(color_string.substr(1, 2),
		nullptr, 16));
	auto g = static_cast<std::uint8_t>(std::stoi(color_string.substr(3, 2),
		nullptr, 16));
	auto b = static_cast<std::uint8_t>(std::stoi(color_string.substr(5, 2),
		nullptr, 16));
	return {r, g, b};
}

void Backlight::SetColorInternal(std::fstream& stream, Color color)
{
	std::ostringstream temp_stream;
	temp_stream << '#' << std::hex << std::setfill('0');
	temp_stream << std::setw(2) << static_cast<int>(color.red);
	temp_stream << std::setw(2) << static_cast<int>(color.green);
	temp_stream << std::setw(2) << static_cast<int>(color.blue);
	//throw std::runtime_error{temp_stream.str()};
	stream << temp_stream.str();
	stream.flush();
}

}
```

I've made a small Qt GUI with this and it works fine.