MatMoul/g810-led #187: Makefile issues

justinkb, 2019-09-17:
This project's makefile has many issues, I'll name a few:

1. doesn't honor PREFIX anywhere, even though it's specified in it for some unknown reason
2. doesn't honor DESTDIR everywhere, for example, on line 86 it should say $(DESTDIR)/etc/ twice instead of /etc/.
3. you probably don't want to symlink to full paths if all the those links are gonna reside in the same directory as the actual g810-led executable anyway

MatMoul, 2019-09-20:
Actually the make file was edited from 7 contributors and me...

If you have some remark, you can comment the code or if you sure of you, make a PR (read https://github.com/MatMoul/g810-led/blob/master/CONTRIBUTING.md).

MatMoul, 2020-04-26:
Seem an old issue

closed 2020-04-26 by MatMoul