MatMoul/g810-led #198: Support for G915

lukaszuznanski, 2019-11-15:
Is there any way to run led controller for g915? "Matching or compatible device not found !"

MatMoul, 2019-11-18:
Hi, g915 aren't supported now, take a look at #188 
Try your previous g810-led commands with sudo.

VictorRogers, 2020-02-17:
@lukaszuznanski Did you ever get this working? I tried running your commands as sudo with no luck

yawor, 2020-03-10:
I'm also trying to make this work on my G915. First I've built it normally and when I run:
`sudo g810-led -dv 046d -dp c541 -tuk 1 -a 000000`
I got `Error: Can not write to hidraw, try with the libusb version`

So I've recompiled it with libusb option and now when I run it in the same way as above, I don't get any message, but the return code is `1` and nothing changes on the keyboard. I've also tried with `--tuk` values 2 and 3.

yawor, 2020-03-13:
After doing some captures I've found out few things. The G915 (or the wireless dongle actually, identified by vid/pid `046d:c541`) has three interfaces in USB descriptor, each containing one HID endpoint:
Interface 0 - Endpoint 0x81 - keyboard input
Interface 1 - Endpoint 0x82 - mouse input (used for multimedia keys)
Interface 2 - Endpoint 0x83 - used for all the G-features (LEDs, configuration etc).

The g810-leds has two modes of operation: hid and libusb.
In libusb case, the problem is visible right away: it assumes the keyboard uses interface 1 and endpoint 0x82 for communication, which is going to fail right away for G915, as it uses interface 2 and endpoint 0x83.
In hid mode, the code opens first hidraw interface it finds with given VID/PID and optional serial number (that's how `hid_open` function works; in case of G915, the serial number is unusable, because the dongle doesn't provide one). This fails because it tries to open interface 0 (I'm wondering how this works with other keyboards). For G915, the `LedKeyboad::open` should find all matching hid devices and filter by interface number (must be equal to 2), get the device path and use it with `hid_open_path` function instead.

**LED control**

There are at least 2 different commands to set LED colours
1. Set up to 4 LEDs, each with its own color:
```
11 01 0b 1d [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB] [AA] [RR] [GG] [BB]
```
where AA is a led address and following it is the RGB value for that LED.
If less than 4 LEDs are being set, then first non-used AA value is set to 0xFF and the rest of the data is filled with 0x00. For example:
- set 4 multimedia keys to color `1400ff`:
```11 01 0b 1d 9d 14 00 ff 9e 14 00 ff 9b 14 00 ff 9c 14 00 ff```
- set backlight level button to color `707070`:
```11 01 0b 1d 99 70 70 70 ff 00 00 00 00 00 00 00 00 00 00 00```

Keyboard responds with
```
11 01 0b 1d [AA] [AA] [AA] [AA] 00 00 00 00 00 00 00 00 00 00 00 00
```
where AA are addresses of LEDs being set (positions not used are set to 0xFF).

2. Set up to 13 LEDs to the same color:
```
11 01 0b 6d [RR] [GG] [BB] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA] [AA]
```
where first 3 bytes are RGB value, followed by 13 LED addresses (if less, then unused positions are set to 0x00).

Keyboard responds with:
```
11 01 0b 6d [RR] [GG] [BB] [AA] 00 00 00 00 00 00 00 00 00 00 00 00
```
where RGB is the same value as set in the command and AA is an address of the first LED in the command (I wonder if this is some bug - my keyboard is running on newest firmware).

3. Commit color change:
```
11 01 0b 7d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
Keyboard responds with the same data

LindaLawton, 2020-05-01:
@yawor  did you get this working?  The issue is g915 but it closes g815 im confused.

Im still having no luck getting this to work with my G915 Im really tired of the random light show or no lights at all.

yawor, 2020-05-01:
@LindaLawton the G915 is identical to G815 regarding the protocol, except for the second byte in the request. This library (and keyleds too) sets this byte to 0xFF, which is fine for wired keyboards, as wired keyboards ignore this value. You can test this by unplugging the wireless receiver for G915 and plugging it with a USB cable to your PC. It'll work exactly like G815.

The problem with wireless keyboard like G915 is that, the second byte is not ignored anymore. It's used to select a device the software is talking to - this is because both the keyboard and the dongle are using Logitech's HID++ protocol. If the second byte is equal to 0xFF, then the software is not talking to the keyboard, but to the dongle - and obviously it doesn't work. The software needs to talk to the keyboard over the wireless link. To do that, the second byte needs to be equal to 0x01 in case of G915 and LightSpeed link.

The LightSpeed dongle is very similar to a Unifying one regarding the HID++ protocol usage. In case of Unifying the second byte does the same exact thing as LightSpeed, but values can be from 0x01 to 0x06 for up to 6 paired devices and 0xFF for the dongle itself. LightSpeed doesn't have pairing ability, so it only supports two values (0x01 for the keyboard and 0xFF for the dongle).

yawor, 2020-05-01:
If you look anywhere in the code where the request packet is created, you'll see something like:
https://github.com/MatMoul/g810-led/blob/master/src/classes/Keyboard.cpp#L383
```cpp
data = { 0x11, 0xff, 0x0c, 0x5a };
```
What I mean is the second value, which is 0xff here and everywhere else.

LindaLawton, 2020-05-01:
>unplugging the wireless receiver for G915 and plugging it with a USB cable to your PC.

I have it plugged into a dock could this be the issue.  Should i try pluging the wire directly into the laptop?

I did try with the g815 commands and they failed.   [#233](https://github.com/MatMoul/g810-led/issues/223)

yawor, 2020-05-01:
Try changing the USB PID from c33f to c33e everywhere in the code and udev rules to test it as wired.

LindaLawton, 2020-05-01:
I only found two places in the code with c33f  i changed them as you mentioned.

ran the following again.

Linda@linda-ThinkPad-X1-Extreme-2nd:~/g915/g810-led$ sudo make install-dev
Llinda@linda-ThinkPad-X1-Extreme-2nd:~/g915/g810-led$ sudo make install-lib
Llinda@linda-ThinkPad-X1-Extreme-2nd:~/g915/g810-led$ sudo make install-dev

I am not actually sure its comping my changes though.

yawor, 2020-05-03:
@LindaLawton actually I'm also having issues trying this library to get to work, even though it should with the changes I've described above. It doesn't work with either hidapi or libusb. You can check G915 branch in my fork (https://github.com/yawor/g810-led/tree/g915) to see if it works for you. I had a strange behaviour where the keyboard stopped working at all and I had to turn it off and on using power switch and even use a different keyboard to disable g810-leds to get it working again (this was on libusb).

I've also been playing around with keyleds library and there I was able to get the communication with the keyboard going (over hidapi), but the current version and structure of the library doesn't allow easy implementation of led control with G815/G915. I've been discussing this with the author of keyleds and he has some plans how to make it more flexible, but I think the main issue is the lack of time.

I've also been writing my own tests using hidapi in Python and I've been able to control LEDs on my G915. If I have some time I could post it somewhere for you to play around, but these are just scraps of code put together quickly to test if it works. I've actually started planning my own application in Python for these new keyboards, but it's not on my priority list right now.

Regarding the protocol itself, it's a proprietary Logitech HID++ protocol (version 4.0 or 5.0, I don't remember). There's some documentation laying around the Internet, but most of it is outdated. The protocol is modular - on the hardware (like keyboard, mouse or game controller) there can be multiple services and the device can return IDs of these services and a value which is then used to access their APIs. G815/G915 keyboards actually share most of the services with older keyboards, but LED and effect controls are newer ones.

LindaLawton, 2020-05-03:
I have been trying to use the Hub from windows and saving this to the board and thats not working either i keep getting this random cycling though the different color presets.  

Its a great board for programming but i really cant take this color changing.   My G11 worked fine.

yawor, 2020-05-03:
@LindaLawton on the keyboard itself, you can access multiple build-in color presets by pressing backlight button + a number key from the main area (not numpad). The build-ins are under 1-7 and 0 (1-7 are different animations, 0 sets everything to Logitech blue). You can also use backlight button + "+" or "-" keys to increase and decrease animation speeds.
There are also two slots (under keys 8 and 9) for user defined color presets. You can store them from the Logitech Hub software. You define it first in the Hub software and then assign the preset to keys 8 or 9 in keyboard settings in the Hub software.
Unfortunately the keyboard doesn't remember the last used stored preset, so if you press BL button + 8 for example and then power cycle the keyboard, it'll switch to default one. But there's also a solution for that. You can change build-in settings for M keys. You can do that also from keyboard settings in the Hub software. If I remember correctly, to change them, you need to switch mode of the keyboard by disabling software management in the settings. After that the Hub software gives you the ability to actually store settings on the keyboard itself. When you do that, you can for example assign your stored color preset (for example BL+8) to be a default preset for M1.

LindaLawton, 2020-05-04:
>backlight button + a number key from the main area (not numpad). The build-ins are under 1-7 and 0 (1-7 are different animations, 0 sets everything to Logitech blue). You can also use backlight button + "+" or "-" keys to increase and decrease animation speeds.

Yes i have that.  But the issue is after about 30 seconds - 2 minutes it changes which color setting its using and I get another random one wither it be rainbow effect or water splash effect you never know its totally random.

I am trying to use this as my daily driver for coding the effects are distracting. Its a keyboard i just want one static color.    I think we will be bringing this back tonight its to expensive of a keyboard to not be working properly.

yawor, 2020-05-04:
Well, it's a "gaming" thing 😏.
Anyway, this is strange. I don't experience any randomness at all. I've set it up once using Logitech Hub software and I'm using it with my own static backlight preset with my Linux laptop (which is my main machine). I don't have any special software installed, it's just what's set in the HW.
Btw did you upgrade the keyboard's firmware? I upgraded mine in Hub when I connected it to a Windows machine for the first time.

LindaLawton, 2020-05-04:
>Well, it's a "gaming" thing 😉.

I play games to but dont need a light show.  Think its an age thing.   😉

Yeah it updated on windows. I over wrote all the M1 -M3 as well as brightness 0 and 9.   When it goes into sleep mode now it comes with a light show. Once i touch it it seams to remain static until i stop typing again.

So what your saying is that i should be able to configure it from windows to stop the light show? and it will remain static in Linux?

I was starting to wonder if it was something in Rider or Webstorm that was setting off that extra bit.

yawor, 2020-05-04:
I only get the light show when I explicitly select it with backlight button + one of the number keys. Other than that it's always static. It's static when I power it on and when it goes to sleep and wakes up.

Here are my settings:
FW: 9.1.35

First I've configured everything with ON-BOARD MEMORY MODE: Off

In my Default profile I've changed only G keys to be empty (I'll maybe change them in the future).
My lighting preset is set to Freestyle with my own colors applied to different sections.

Low Battery Mode (Auto):
Breathing Red (Logo Only)

Inactivity Lighting:
Dim Brightness 50%
Start after: 1 minute
All lights off after: 5 minutes

Stored Custom Lighting:
KEY 8: my preset
KEY 9: not set

ON-BOARD MEMORY MODE: On
All M slots are set to Default profile with Lighting set to lighting preset stored on KEY 8.

mienkofax, 2020-05-15:
Hello, I tried @yawor branch with G915 support but i have problem with set keys, not working, only all leds are disabled. Have you a small example, that set led of key? For example Python or C++
```
open usb
write, set led of key 
close
```

yawor, 2020-05-15:
@mienkofax I've tried porting this quickly. It doesn't work for me either. Maybe I've missed some parts of code that also need to be changed. It needs debugging. My fork can be used as a base to work further on this.
I don't have access to the keyboard until Tuesday.

MatMoul, 2020-05-15:
For info, I've made a small tool named `rawwrite` in led-g810-resources (hidapi only) :
https://github.com/MatMoul/g810-led-resources/tree/master/tools/rawwrite
The tool help a lot for testing :
`sudo ./rawwrite vendorID productID rawdata`
as ex :
`sudo ./rawwrite 046d c541 11ff107c00000000000000000000000000000000`

mienkofax, 2020-05-15:
Thank you for info @yawor. @MatMoul this tool is very very helpful. I tried it, but I have a problem writing to the hid `hid_write()` -> `Error: Can not write to hidraw`

I captured communication (https://github.com/mienkofax/keyboard):
 * disable all leds
 * disable leds on the arrows
 * enable leds on the arrows
 * enable led for key `w`

And I tried run tool with:
```shell
./bin/rawwrite 046d c33e 10ff0e3c000000
sleep 0.001
./bin/rawwrite 046d c33e 10ff0e1c000000
sleep 0.001
./bin/rawwrite 046d c33e 10ff142c000000
sleep 0.001
./bin/rawwrite 046d c33e 10ff131c000000
sleep 0.001
./bin/rawwrite 046d c33e 10ff131c000000
```
But after all commads: `Error: Can not write to hidraw`

Example: disable all leds
First command from Logitech Ghub, wireshark:
[image]

I send: `./bin/rawwrite 046d c33e 10ff0e3c000000`

My submitted data, captured request using wireshark
[image]
[image]

And response:
[image]

I'm a beginner in this domain, sorry, but I want to help. I think it will be just a trifle, e.g. padding or maybe typo.

Thank you very much for any help.

MatMoul, 2020-05-15:
@mienkofax `sudo` ?
Or run `rawwrite` as root...

mienkofax, 2020-05-16:
@MatMoul, yes I run with sudo and I tried run under sudo user, nothing helped. Data was probably sent, wireshark capture, but maybe message format is not correct.

G915 supports two types of connection:
 * usb cabel: `Bus 001 Device 023: ID 046d:c33e Logitech, Inc. G915 WIRELESS RGB MECHANICAL GAMING KEYBOARD`
 * wireless - Lightspeed: `Bus 001 Device 024: ID 046d:c541 Logitech, Inc. USB Receiver`

MatMoul, 2020-05-24:
@mienkofax Thanks for the infos...
It seem that the g915 need to be adressed with these 2 differents product ID...

GitJamz, 2020-05-30:
Hi all.
Im no coder whatsoever. Im just a linux user (beginner in linux) that want to use this keyboard with my manual colors.
The only way is the way described above right? To set it up in windows according to @yawor instruction couple posts up. Did anyone else try this yet?
I have the G915 right now but Ive pro-ordered the 915 TKL keyboard that will arrive in a week.. I will try this out then.
Right now I have the exact same issue as @LindaLawton . I can press lightkey + 0 and it becomes static for a little while. Then it begins its lightshow again..

yawor, 2020-05-30:
I don't know if @LindaLawton managed to duplicate what I did, but I hope that she did.
I did my setup with Windows 10 running in VirtualBox - just attached USB receiver to Windows VM and Logitech Hub software detected it without any issues. I'm using G915 with my Linux laptop all the time as my daily driver - no light show at all - only my own static lighting 100% of time.
I also have a second PC with Windows where the Hub software is installed. My monitor has a KVM built-in so it switches USB uplink between laptop and PC depending on the video input selected. I've set up a little different color profile in Logitech Hub so I get different colors depending on the system I'm using it with.

LindaLawton, 2020-06-02:
I have not been able to get this to work.  The closest i came was setting it up in Windows to stay a static color but when it goes into sleep mode i still get a random light show which is slightly annoying but i have learned to live with.  

None of the commands i have tried with this library have done anything except say its not supported.

yawor, 2020-06-02:
@LindaLawton either I'm lucky or there's something wrong with your keyboard.
What I can confirm is that, when the keyboard goes to sleep, it always goes back to a color schema assigned to active M key. So when I change schema to built-in all blue one by hand and wait for keyboard to go to power save mode and press something, it goes back to my color profile. It uses my saved profile even when I unplug the USB receiver.

BTW I'm not using any software control under Linux. I've set it up once on Windows and stored everything in keyboard's built-in memory. I can create a video and post it somewhere.

LindaLawton, 2020-06-03:
@yawor I have also tried to set it up under windows.  Any chance you want to take a picture of your settings in the Logitech Hub?   I am having a heck of a time to get it to save the the keyboard there are no buttons in the UI, like "SAVE" or "STORE".  I'm issues understanding the UI.  I even have one key that wont light up under Linux but when I am in windows it has light.

yawor, 2020-06-03:
@LindaLawton sure

[image]

Here you first configure your profile and name it (here it's saved under a `YAWOR 1` name. Then click on the settings icon (marked green).

[image]

On settings page, check your power saving options (marked blue; I don't know if they have any effect when in on-board mode though).

Then assign your saved backlight profile as stored custom lighting to either one of KEY 8 or KEY 9 (yellow box).

Then change mode to on-board mode (green box).

[image]

In this mode you can assign a keyboard profile (this is mapping of G keys; green box) and lighting profile (yellow box) to each of the M keys. I don't know if changing keyboard profile is needed, but if you change both, then first set the keyboard profile and then lighting profile, as keyboard profile overwrites lighting profile (at least for me).

Here's the final effect:
https://www.youtube.com/watch?v=LsgbMbX0phY
The keyboard is not plugged in (the receiver is visible above F keys). As you can see, it uses my profile from the first screenshot right after turning it on.

MattSturgeon, 2020-08-08:
Looking at [this comment](https://github.com/MatMoul/g810-led/issues/188#issuecomment-599037369) it seems now that the G815 is supported, G915 support should be fairly trivial. Sounds like just the device identifier byte needs to be set in the wireless protocol? Is anyone looking into this?

LindaLawton, 2020-08-11:
I wish someone would :)

mcdope, 2020-08-11:
Based on the comments from mienkofax and matmoul it seems to be not as simple as changing the device identifier.

If all that's missing is some id-shuffling and minor code tweaks to make the G815 parts work with the G915 I would be happy to submit a PR since I use that keyboard too and the idle-animation annoys the shit out of me. 

But that would require someone pointing me into the right direction. Based on the comments above it seems to be not possible at the moment as far as i understand.

MatMoul, 2020-09-03:
I've just received a G915...
I can now work on it...
Be patient

MatMoul, 2020-09-14:
I did some tests, but it doesn't seem so simple ...

yawor, 2020-09-15:
@MatMoul Yes, I've tried to apply what I've said in my previous comments but for some reason I'm not able to access the G915 from this app. You can check my attempt here https://github.com/yawor/g810-led/tree/g915
I'm able to successfully send HID++ commands to the keyboard (in both wired and wireless mode) using libusb instead of libhid from a quick and dirty python scripts I've written. I've tried doing it with libhid, but I wasn't able to do it.

mienkofax, 2020-09-17:
@yawor could you publish a python script on GitHub? It would be amazing. :slightly_smiling_face:

yawor, 2020-09-17:
@mienkofax this is nothing fancy, it's just a simple script. For example, this just sets everything to white:

```python
import usb.core
import usb.util
import usb.control

dev = usb.core.find(idVendor=0x046d, idProduct=0xc541)  # type: usb.core.Device
cfg = dev.get_active_configuration()
#print(cfg)

if dev.is_kernel_driver_active(2):
    dev.detach_kernel_driver(2)
usb.util.claim_interface(dev, 2)

try:
    while True:
        print(dev.read(0x83, 20, 1))
except usb.core.USBError:
    pass

# Set 3 LED ranges to white:
# 0x01 - 0x6f: main area + f-keys + arrows + ins/del/etc + numpad
# 0x9b - 0x9e - multimedia buttons
# 0xb4 - 0xb8 - G-keys
dev.ctrl_transfer(0x21, 0x09, 0x0211, 0x02, bytearray([0x11, 0x01, 0x0b, 0x51, 0x01, 0x6f, 0xff, 0xff, 0xff, 0x9b, 0x9e, 0xff, 0xff, 0xff, 0xb4, 0xb8, 0xff, 0xff, 0xff, 0x00]))
print(dev.read(0x83, 20))

# Set 2 LEDs to white:
# 0x99 - backlight level button
# 0xd2 - G logo
dev.ctrl_transfer(0x21, 0x09, 0x0211, 0x02, bytearray([0x11, 0x01, 0x0b, 0x11, 0x99, 0xff, 0xff, 0xff, 0xd2, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]))
print(dev.read(0x83, 20))

# Commit color changes
dev.ctrl_transfer(0x21, 0x09, 0x0210, 0x02, bytearray([0x10, 0x01, 0x0b, 0x71, 0x00, 0x00, 0x00]))
print(dev.read(0x83, 20))
```

I don't know if it works in Python 2.7, I'd recommend Python 3.5 or newer. It uses pyusb library. You need to execute this as root (for example, using sudo).

mienkofax, 2020-09-23:
@yawor thank you very much for your support. It works perfectly. I will share my code.

GitJamz, 2020-10-14:
@yawor @LindaLawton MattSturgeon 
Hey guys/gals,

Any updates about the H915 support on linux? I jut want to be able to control the lights but unsure what the state is on this project.. Someone care to clarify where we are at?

GitJamz, 2020-10-19:
Ok so.. How do we solve this.. Whoever can fix this let us know your donation goal to get it done.. I want to donate for this feature and Im sure a few others wanna do as well..

louis77, 2020-10-19:
@GitJamz I'm willing to donate to. 

However, Logitech is to blame to release hardware without proper Linux support. At least they could have release a CLI for minimum config.

MatMoul, 2020-11-06:
Thanks for all the samples but I'm trying to use hidapi because it is faster than libusb.
But it seems that this is not possible for the g915.
I'm very busy but if I find time I'll work on it ...

GitJamz, 2020-11-06:
> Thanks for all the samples but I'm trying to use hidapi because it is faster than libusb.
> But it seems that this is not possible for the g915.
> I'm very busy but if I find time I'll work on it ...

Please man.. consider giving us a donationgoal so that we can donate for this!

Elijas, 2020-12-16:
@yawor Could you create a separate repository (**Not** a GitHub fork) with copied over code (the license allows this) and then you could also create a ([ko-fi.com](https://ko-fi.com/)) account specifically for the new repository so we could fund (or at least show support) for the effort

yawor, 2020-12-16:
@Elijas thanks for the offer, but I don't have time to try to port this to g915 by myself. I've tried as much as I could to help add support for g915 to this project, but I'm not that fluent in C/C++ to make bigger changes. In my opinion, the current architecture of this library is not optimised for supporting different keyboards - it would need quite a big refactoring (for example a base keyboard class with classes for each keyboard model inheriting from it).
To be honest, I like more the approach of keyleds project, where it doesn't differentiate features based on the keyboard model (USB PID), but actually requests a list of features from the HID++ devices themselves - this is a much more future-proof approach. Unfortunately it doesn't support the new LED API available in g815/g915 yet.

Elijas, 2020-12-16:
@yawor I see, thanks for letting us know and for your contributions so far. Let's hope the support will be there at some point one way or another :)

@louis77 @GitJamz I'd invite posting the request to [r/LogitechG](https://www.reddit.com/r/LogitechG), as Logitech's official customer support is tracking the requests there

Eosis, 2020-12-22:
The two relevant posts on r/LogitechG that I found were:

[Here](https://www.reddit.com/r/LogitechG/comments/jq4ckp/g915_tkl_linux/) and [here](https://www.reddit.com/r/LogitechG/comments/kfj8o5/support_of_g915_on_linux/).

It definitely doesn't feel like it is a priority. I'm not sure how big this market is but I can't imagine it is very large, so it will probably  remain off their radar. 😓

mcdope, 2020-12-22:
Logitech has never cared about customers, I wouldn't expect ANYTHING from them.

I had two contacts with Logitech. Once I've asked for a spare part to fix my G15 (the flat cable, connecting keyboard and display), response: send in the keyboard, we replace it and only charge your firstborn for it. The second: asking for some other spare part - replacement keycaps for some other mechanical gaming keyboard which had WASD worn off after like 20 months. Response this time: sry, dont have em. But send in the keyboard, we will replace it and only charge....

Considering this experiences, and the user base of Linux, well... do the math :D

Elijas, 2020-12-22:
> Logitech has never cared about customers, I wouldn't expect ANYTHING from them.
>
> I had two contacts with Logitech. Once I've asked for a spare part to fix my G15 (the flat cable, connecting keyboard [...]

Given hundreds of different products being sold concurrently around the world and many more retired, where each product itself has many dozens of parts, managing the logistics of parts is quite difficult/expensive to do.

On the other hand, releasing just the barebones MIT-licensed (for no warranty, to save up on costs for development) library on github to allow scripting would cost nothing

molecula21, 2020-12-27:
Any news on any way to get one king of control from linux on this? Just got mine and never thought on checking for linux compatibility. It is a keyboard after all. Now all this light show after the keyboard goes to sleep is just not usable at all.

Elijas, 2020-12-27:
> Any news on any way to get any kind of control from linux on this? Just got mine and never thought on checking for linux compatibility. It is a keyboard after all. Now all this light show after the keyboard goes to sleep is just not usable at all.

#### How to fix the rainbow lightning effect rest after sleep
Install G HUB (Windows or MacOS), connect keyboard with Lightspeed, go to Settings, and set `ON-BOARD MEMORY MODE` to `ON`. This is a one-time setup for the lifetime of a keyboard.

#### Other features that work on Linux
Works out of the box:
- All the media buttons, game mode (disables the <kbd>Win</kbd> key), volume control
- Pressing <kbd>☀</kbd> cycles through brightness levels.
- Pressing <kbd>☀</kbd> + <kbd>[NUM]</kbd>, where <kbd>[NUM]</kbd> is <kbd>0</kbd>, <kbd>1</kbd>, <kbd>2</kbd>, ..., <kbd>9</kbd> allows you to change the lightning effect on the fly.

Requires one-time setup with G HUB (connected through Lightspeed), then works everywhere you attach the keyboard to, including Linux:
- You can create custom animations, per-key lightning (or download free animations created by others in the animation "app store") and then store two of them on the keyboard itself. The animations are accessible through <kbd>☀</kbd> + <kbd>8</kbd>, <kbd>☀</kbd> + <kbd>9</kbd> (You can also set it to <kbd>FN</kbd> + <kbd>F1</kbd>, <kbd>FN</kbd> + <kbd>F2</kbd>, etc.)
- You can change the keyboard sleep timeout

molecula21, 2020-12-27:
Thanks, I was able to set it on windows and using your instructions plus @yawor screenshots. It now stays as I have setup in GHub even after the keyboard goes to slepp.

Eosis, 2020-12-29:
I was playing around with this and hit upon a solution that is working for me. My main goal was preventing the color-wave effect that is seen on the g915 if it goes to "Sleep" on memory mode (which occurred to me after one minute of inactivity).

It is a little unclear to me if this is what other posters here have experienced. I think it is closest to the issue that @LindaLawton described. My settings configured to a memory slot in ghub would persist on the keyboard, but only if I'd pressed a key in the past minute. After this minute passed, I would get a rainbow wave which I found distracting.

I actually ended up installing the libratbag daemon (installation from package managers [here](https://github.com/libratbag/libratbag/wiki/Installation), though I built it from source), then setting the led's through `ratbagctl`. There are only two leds there, one that is just the logitech logo and another for the rest of the keys. I set both to the same color through this utility, and now I don't get the rainbow wave after 1 minute. I also needed to connect the keyboard via wire to my machine for `libratbagd` to pick it up.

I'm not so sure this is very robust, so YMMV, but it has worked for me and hopefully will help others.

yawor, 2020-12-29:
@Eosis were you trying to follow steps I've described [here](https://github.com/MatMoul/g810-led/issues/198#issuecomment-638111800)? In GHub, you need to store your own color profile/animation under KEY 8 or KEY 9 and then you need to also assign the stored profile to M1, M2 and M3 buttons in ON-BOARD MEMORY MODE.
I have my own profile stored and assigned to all three M buttons and my profile (static colors, no rainbow) is activated automatically on keyboard power on and also on wake up after sleep.

mcdope, 2020-12-29:
> @Eosis were you trying to follow steps I've described [here](https://github.com/MatMoul/g810-led/issues/198#issuecomment-638111800)? In GHub, you need to store your own color profile/animation under [...]

I've did exactly that, works fine for my G815 (at home) - but not for the G915 (using Lightspeed dongle, at work). The G915 still always activates that rainbow stuff if you don't use it for a while.

yawor, 2020-12-29:
@mcdope that's really strange. Maybe G915 has different hardware revisions? Mine is G915 (not G815) and after setting it once in GHub I've never seen it using the rainbow effect on power on/wake up. I'm using it always in Lightspeed mode (wireless). I can activate rainbow effect manually using backlight button + 1, but after it goes into low power mode, when I press anything it goes back to my own profile.
No software installed on my system at all which would do that (actually I can even unplug the receiver and it still behaves correctly).

mcdope, 2020-12-29:
@yawor wierd, will re-set my profiles when I'm back in office (got the laptop here, but not the keyboard :smile: ) and see if that changes something. Maybe it was a bug in earlier versions of Ghub... Who knows

GitJamz, 2021-01-02:
I managed to set my G915 TKL according to the instructions above with setting my profile on the onboard memory.
Now it seems to stay this way always! However as soon as it goes to "inactivity" mode it goes back to this ridicolus breathing rainbow in random color until I press any button. Then its back to my saved profile again.

I have set the inactivity energy saving setting to dim to 50% but it does not dim - just goes to rainbow..

Elijas, 2021-01-03:
For it to not set to rainbow you need to enable "on-board memory" mode as
well

[27 more replies]