MatMoul/g810-led #199: "Matching or compatible device not found !" for G910 Orion Spectrum.

Galbi3000, 2019-11-20:
I am tired of seeing the constant colour changing on my Logitech G910 Orion Spectrum keyboard as the shift from green to blue to red makes it look like the keys are moving and it bugs me while I am typing (I am not a touch typist!).

I have a dual boot system with Windows 10 and Kubuntu Linux 19.04 64-bit. The Linux drivers do not have the full functionality that the Windows ones do which is a great shame on the Linux community! I needed to take control of the keyboard LEDs just like can be done in Windows and stop the annoying rainbow cycle effect!

I came across g810-led in a Google search, as far as I could find it is the only software that can do what I want. When I looked into it further I found it had support for my model of keyboard built in which had worried me due to the software being titled after the G810 keyboard.

So my problematic journey into trying to get g810-led to work began. Here is how it went:

- I followed the instructions in INSTALL.md to the letter and got no errors. I did notice you missed sudo in the instructions for Ubuntu, I'm sure this is just a simple oversight as most people know you need root privileges in Ubuntu to install anything :)
- I tried running g810-led I was told "Matching or compatible device not found !". The keyboard is plugged directly into the PC and not through a USB hub (I always do that as the BIOS does not like it on a hub for getting into the BIOS settings!)
- I thought maybe the distro version is not the latest one so I decided to compile from source by following the instructions further down INSTALL.md (I.E. "git clone https://github.com/MatMoul/g810-led.git" etc.). I got no errors during the make stage.
- I tried running the generated executable and was again told "Matching or compatible device not found !".
- I read in INSTALL.md "hidapi is recommended but if you encounter a problem on your system, switch to libusb" so I did that with "make bin LIB-libusb", I also did a "make clean" first to be sure it was recompiling properly. This also did not produce any build errors but still gave the same "Matching or compatible device not found !" error at run-time.
- I then tried running with the --list-keyboards option and got this error: Segmentation fault (core dumped)

I have not tried the steps in the "Testing unsupported keyboards" section of README.md as I felt I should not have to do that when my exact keyboard is listed in the supported keyboards.

So here I am looking for help! I hope my 'story' is detailed enough to get some idea of what is at fault. My dated programming expertise is rather rusty and focused more on GUI than drivers so I did not look further into the issue by looking at code that I would likely not fully understand.

From what you said in the other thread I have a suspicion that the segmentation fault is a separate issue from the "Matching or compatible device not found !" one.

I do not use my Linux installation all the time, I am mostly in Windows for gaming, so getting further info for you or trying suggestions will likely take longer than usual.

MatMoul, 2019-11-20:
Thanks for your isssue,

It seem you have 2 issues...
"Matching or compatible device not found !"
"Segmentation fault (core dumped)"

"Matching or compatible device not found !"
Firstly try to do a `lsusb`, you will find your vendor and product ID (046d:????)
After that you can try to take a look if your product ID is in the code : https://github.com/MatMoul/g810-led/blob/118631d184e18fe8ba2d233bd5e059fea9e621b6/src/classes/Keyboard.h#L47
If your product ID is in the code, your keyboard is supported, if not you can try unsupported keyboard.

"Segmentation fault (core dumped)"
You're the first with a seg fault, I suggest we try to resolve the first case before this.

Sudo or not sudo :
When you test unsupported keyboard, use sudo.
But for supported keyboard, if you have copied g810-led.rules in udev, you don't need to use sudo

Galbi3000, 2019-11-20:
I'm not in Linux today so I'll have to look tomorrow (the game I am playing is in Linux but the mouse movement feels too odd in Linux, the DPS is too high and like the keyboard drivers the mouse drivers are not as featured as in Windows, can not adjust the mouse DPS! Who says Linux is better? :P )

The sudo comment is about the command "apt install g810-led" this should be "sudo apt install g810-led" :)

Galbi3000, 2019-11-23:
OK, had a bit of trouble with Linux after I upgraded my video card! (Again, who says Linux is better? No trouble using Nvidia's drivers in Windows! :P )

I just installed g810-led using the Ubuntu repository as I do not have the project downloaded currently due to re-installing Linux :/

lsusb returns the following line for the keyboard:
```
Bus 005 Device 002: ID 046d:c335 Logitech, Inc. G910 Orion Spectrum Mechanical Keyboard
```

0x046d, 0xc335 is listed in your keyboards.h file, line 57.

So I ran g810-led and I got the following result:
```
~$ g810-led -a ffffff
Matching or compatible device not found !
```

The following worked:
```
sudo g810-led -dv 046d -dp c335 -tuk 2 -a 000000
```

So I tried the first attempt again but using sudo and it worked!
So it seems I have to use root priviledges all the time.

Oh, I tried the --list-keyboards again with the repository version and got this result:
```
~$ g810-led --list-keyboards
Device: Logitech - Gaming Keyboard G910
        Vendor ID: 046d
        Product ID: c335
        Serial Number: 0873364D3435
```
Notice I did not use sudo there but it saw the keyboard!

Galbi3000, 2019-11-23:
So, from my perspective the program is working to a degree, as long as I use sudo. However, there is the segmentation fault I got when I tried making the project the first time I had troubles.

Even though it is working for me using the repository version (Version : 0.3.3) I may still take a time out at some point and see if I can replicate that for you.

MatMoul, 2019-11-24:
@Galbi3000 I'm not the packager of the deb package, do you have this file https://github.com/MatMoul/g810-led/blob/master/udev/g810-led.rules in /etc/udev/rules.d/ ?
This file allow you to use g810-led as user.

For the segmentation fault, no idea, it's the first case in 3 year, maybe a system not completely updated during the construction of g810-led...

Holger-Will, 2020-02-02:
i just had the exact same problem... I have a g810 (c337) I'm on ubuntu 18.10, build from source with `libusb`, after make install the keyboard profile got loaded and applied so it did seem to work. 

but when i tried to do anything with `g810-led` i get 

`Matching or compatible device not found !`

and on `--list-keyborards` i got SEGFAULT.

using sudo with `list-keyboards` worked, but without sudo: SEGFAULT...

after a reboot `--list-keyboards` did not produce a segfault anymore...

**fix**
then i tried rebuilding with `hidapi` and everything works great. no segfault, no reboot required and setting colors works perfect!
Thanks for the great tool!

closed 2020-04-26 by MatMoul