MatMoul/g810-led #200: G512 Special Edition - Matching or compatible device not found

odziom91, 2019-11-30:
Hello,
I bought G512 Special Edition keyboard. After installation g810-led-git (package from AUR in ArchLinux) and running with parameters --list-keyboards and --print-device I received information:
Matching or compatible device not found !
and the same info with sudo.
I tried to testing my keyboard. I put lsusb: 046d:c342 and test keyboard with command:
g512-led -dv 046d -dp c342 -tuk 1 -a FF0000
Keyboard changed color to red so it is ok. How to make this keyboard working without additional -dv -dp and -tuk params?

Thanks in advance!

_Originally posted by @odziom91 in https://github.com/MatMoul/g810-led/issues/138#issuecomment-560009277_

odziom91, 2019-11-30:
After a little investigation in source code I forked original code and add few lines - on my risk. :)
After compilation I can check keyboard with --list-keyboards and --print-device parameters. I tried to change color and run some fx'es with success. So I guess G512 Special Edition should be added to source code.
@MatMoul please check my forked code: https://github.com/odziom91/g810-led.git
I'm not sure that it's okay but feel free to check and use.
Maybe little information about this keyboard - it is a "newer version" of G413 but with RGB lights.

borisoni, 2019-12-11:
Would be great to see this merged! @odziom91, thanks for the effort, used your repo for now.

MatMoul, 2019-12-14:
@odziom91 Thanks for your work, It seem ok for a PR.
PS : Try to work on the develop branch for the next time.

PapiCZ, 2019-12-24:
Thank you! It works great.

wiccan2, 2020-04-14:
@odziom91 Why did you never create a PR for this?

I've installed the modified version on my Arch laptop by changing the PKGBUILD to point to your git repo instead of @MatMoul's and it works great.

I don't mind lending a hand if there's anything more that needs doing to get a PR put in place.

It would be great to see this integrated.

MatMoul, 2020-04-28:
The latested release V0.4.0 include these change...

closed 2020-04-28 by MatMoul