MatMoul/g810-led #201: Writing a script to utilities FN key with F1-12 keys

AhmedBHameed, 2019-12-04:
Hello Sir,

Thank you for the nice library. I have a G513 logitech keyboard. I have a question about if there a possibilities to write a script that execute a specific task OR to type a string when ever the right keys called (which is nice to have really).

As the standard library support programming the F functions keys and bind it with other tasks or for example to print something on screen.

Hope that you get the idea.

Thank you in advance.

MatMoul, 2019-12-14:
g810-led is designed as a command line, not a daemon.
Take a look at https://github.com/CReimer/g910-gkey-uinput

AhmedBHameed, 2019-12-15:
Hmmm, I see. Well still nicely done with colors.
Thank you.

closed 2019-12-15 by AhmedBHameed