MatMoul/g810-led #205: G513 Linear Carbon, G502 Lightspeed (mouse). G935 Headset support

David-Frick, 2019-12-21:
Hey there, great library! Really hoping to use it with my Debian rig.

It seems the G513 Linear Carbon is not detected for some reason. Any advice? Is there a procedure to dump some data from the command line to help? I can run any commands! I use Debian 10 currently.

Also, could this repo but used to set the RGB for my logitech mouse and logitech headset as well? I really hope so! That would be sweet. They are both wireless but do use usbs to talk to the mobo if that helps.

Hoping you are having a good holiday season,
David

benutzer74, 2020-03-08:
I have a G513 tactile and it works with g810-led ver 0.3.9.

MatMoul, 2020-04-28:
Is it working or not ?

abepetrillo, 2020-04-30:
I think some people get confused and dont try the `-tuk`. the "supported" list implies you shouldn't have to try the unsupported method. I have the G512SE and had to use `-tuk 1`

MatMoul, 2020-05-08:
@abepetrillo Thanks for your feedback, it seem the G512SE isn't supported now...
Can you give me the return of this cmd :
`lsusb | grep -i logitech`

MatMoul, 2020-05-16:
Closing, don't reopen...
If issue isn't solved, reopen a new issue for each case.

closed 2020-05-16 by MatMoul