MatMoul/g810-led #206: G413 Silver - Stop Breathing

thereallinusrg, 2019-12-27:
So I tried to stop the breathing effect of my G413 Silver with the G413 Carbon command on Arch.
Sadly, that didn't work.
Is there a way to add support for the G413 Silver / port the Carbon device profile?
What information would I need to create a G413 Silver device profile?
The constant breathing is slowly driving me nuts.

thereallinusrg, 2019-12-27:
I created a Wireshark dump for the G413 Silver:
[G413-Silver-Dump.zip](https://github.com/MatMoul/g810-led/files/4005926/G413-Silver-Dump.zip)

MatMoul, 2020-04-28:
G413 seem supported, is this issue always relevent?

thereallinusrg, 2020-05-02:
Yes, it seems that only the the G413 Carbon is supported, not the G413 Silver.

MatMoul, 2020-05-08:
Sorry, G413 Carbon is supported but not Silver...
Your keyboard seem to be old...

thereallinusrg, 2020-05-09:
Well, as old as the G413 Carbon. But yeah, oldish

mrmusichead, 2020-05-15:
Having the same issue with the G413 Silver. I tried to modify the profile, and when I set the color to blue and restarted, the breathing stopped. The problem now is that I can't get the LEDs to light back up. I'm on Kubuntu 20.04. Also, I switched to Linux about 2 weeks ago from only using Windows, so I have next to no idea how to test the keyboard. Loving Linux otherwise though!

MatMoul, 2020-05-16:
Seem @mrmusichead has replied :)

closed 2020-05-16 by MatMoul