MatMoul/g810-led #210: Looking for collaborators

MatMoul, 2020-01-16:
Hello everyone,

Sorry for my long silent but I'm very busy...

g810-led is now part of Debian Buster.
I think this is time for this project to have collaborators that can help me to maintain it.

The line of the project :
 - Maintain a solution without deamon (one shot command)
 - Maintain a build library for developers
 - Maintain the speed of execution (< 100ms)
 - Track memory leaks (valgrind)

Selection of collaborators :
 - Confirmed contributors
 - Good GitHub with C++ projects
 - Other, your C++ contributions

I'm open for other proposals...

Desulate, 2020-06-11:
Hello,

I'm not involved with the project but I'm currently using [keyleds](https://github.com/keyleds/keyleds). I was previously using your project but eventually found that keyleds worked better for me with my G610 keyboard.

If your looking for collaborators why don't you try reaching out to the developers of that project since your project here has similar goals. I'm sure there must be stuff you guys can collaborate on.

MatMoul, 2020-08-11:
@Desulate Thanks for your feedback,
keyleds seem a very good project but it has some depencies that I can accept and it is very more intrusive than g810-led...