MatMoul/g810-led #212: Total Linux noob alert

gravesmeister, 2020-02-24:
So I'm new to Linux. I'm on Pop OS. I recently wiped Windows 10 away and now I'm only on Linux. Anyway, I'm used to simply downloading a .exe and double clicking and off you go. So when I comes to git hub as you can imagine I'm totally confused on how to get this software running. There's no .exe. So can someone please explain what I do to get this software installed and running? Even if you can link me to where I can go to find out what to do next would be awesome. I give thanks in advance to anyone willing to help this new and eager Linux user to learn how to get something on Git hub working for me. Thank you for your time and the more I use Linux and learn about it the more I want to learn it.

hk0O7, 2020-02-26:
Hi.
I'm not running Pop OS, but since it appears to based on Ubuntu and uses its package repositories, if you have the latest non-LTS version (19.10) you should be able to install it from them as mentioned in [INSTALL.md](https://github.com/MatMoul/g810-led/blob/master/INSTALL.md#installation-using-repos-), with this command:
```
sudo apt install g810-led
```

gravesmeister, 2020-03-13:
> Hi.
> I'm not running Pop OS, but since it appears to based on Ubuntu and uses its package repositories, if you have the latest non-LTS version (19.10) you should be able to install it from them as [...]

Thank you. I actually just found a How To Geek article that shows that noob like me the basics of how to get started with Git Hub. Thanks for your response. I'm now using Salient which is Arch based. I'm a gamer so I want to make my linux machine a gaming one. So far so good.

MatMoul, 2020-04-28:
@gravesmeister If you're on an arch distro, install the `g810-led-git` aur package...

closed 2020-04-28 by MatMoul