MatMoul/g810-led #213: sometimes on boot or waking from sleep they leds cycle serveral times

jgslade, 2020-03-02:
Sometimes when booting up or waking from sleep the leds will cycle from off to the rainbow wave to my chosen profiles several times, while it's doing this cycling I cannot use the keyboard. 

> [ 2381.844326] input: Logitech Gaming Keyboard G910 Keyboard as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.0009/input/input43
[ 2381.898440] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.0009/input/input44
[ 2381.898684] hid-generic 0003:046D:C335.0009: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input1
[ 2382.005006] audit: type=1130 audit(1583166733.670:137): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[ 2382.005012] audit: type=1131 audit(1583166733.670:138): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[ 2382.009411] Generic FE-GE Realtek PHY r8169-300:00: attached PHY driver [Generic FE-GE Realtek PHY] (mii_bus:phy_addr=r8169-300:00, irq=IGNORE)
[ 2382.122710] r8169 0000:03:00.0 enp3s0: Link is Down
[ 2383.020023] usb 3-1.1: USB disconnect, device number 7
[ 2383.391476] usb 3-1.1: new full-speed USB device number 8 using xhci_hcd
[ 2383.505527] usb 3-1.1: New USB device found, idVendor=046d, idProduct=c335, bcdDevice= 6.02
[ 2383.505530] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2383.505531] usb 3-1.1: Product: Gaming Keyboard G910
[ 2383.505532] usb 3-1.1: Manufacturer: Logitech
[ 2383.505533] usb 3-1.1: SerialNumber: 0C6E384D3833
[ 2383.598959] input: Logitech Gaming Keyboard G910 as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.0/0003:046D:C335.000A/input/input47
[ 2383.655106] hid-generic 0003:046D:C335.000A: input,hidraw0: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input0
[ 2383.658145] input: Logitech Gaming Keyboard G910 Keyboard as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.000B/input/input48
[ 2383.714882] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.000B/input/input49
[ 2383.714965] hid-generic 0003:046D:C335.000B: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input1
[ 2385.835653] usb 3-1.1: USB disconnect, device number 8
[ 2386.006823] r8169 0000:03:00.0 enp3s0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 2386.006835] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[ 2386.161331] usb 3-1.1: new full-speed USB device number 9 using xhci_hcd
[ 2386.275283] usb 3-1.1: New USB device found, idVendor=046d, idProduct=c335, bcdDevice= 6.02
[ 2386.275288] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2386.275291] usb 3-1.1: Product: Gaming Keyboard G910
[ 2386.275293] usb 3-1.1: Manufacturer: Logitech
[ 2386.275295] usb 3-1.1: SerialNumber: 0C6E384D3833
[ 2386.351134] input: Logitech Gaming Keyboard G910 as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.0/0003:046D:C335.000C/input/input52
[ 2386.405092] hid-generic 0003:046D:C335.000C: input,hidraw0: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input0
[ 2386.407827] input: Logitech Gaming Keyboard G910 Keyboard as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.000D/input/input53
[ 2386.464737] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.000D/input/input54
[ 2386.465056] hid-generic 0003:046D:C335.000D: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input1
[ 2388.025536] audit: type=1130 audit(1583166739.690:139): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[ 2388.381281] audit: type=1130 audit(1583166740.047:140): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=ntpd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[ 2388.395652] usb 3-1.1: USB disconnect, device number 9
[ 2388.734513] usb 3-1.1: new full-speed USB device number 10 using xhci_hcd
[ 2388.842527] usb 3-1.1: New USB device found, idVendor=046d, idProduct=c335, bcdDevice= 6.02
[ 2388.842530] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2388.842531] usb 3-1.1: Product: Gaming Keyboard G910
[ 2388.842532] usb 3-1.1: Manufacturer: Logitech
[ 2388.842533] usb 3-1.1: SerialNumber: 0C6E384D3833
[ 2388.908270] input: Logitech Gaming Keyboard G910 as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.0/0003:046D:C335.000E/input/input57
[ 2388.964659] hid-generic 0003:046D:C335.000E: input,hidraw0: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input0
[ 2388.965600] input: Logitech Gaming Keyboard G910 Keyboard as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.000F/input/input58
[ 2389.021304] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.000F/input/input59
[ 2389.021467] hid-generic 0003:046D:C335.000F: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input1
[ 2390.699243] usb 3-1.1: USB disconnect, device number 10
[ 2391.054372] usb 3-1.1: new full-speed USB device number 11 using xhci_hcd
[ 2391.162492] usb 3-1.1: New USB device found, idVendor=046d, idProduct=c335, bcdDevice= 6.02
[ 2391.162496] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2391.162499] usb 3-1.1: Product: Gaming Keyboard G910
[ 2391.162501] usb 3-1.1: Manufacturer: Logitech
[ 2391.162503] usb 3-1.1: SerialNumber: 0C6E384D3833
[ 2391.243953] input: Logitech Gaming Keyboard G910 as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.0/0003:046D:C335.0010/input/input62
[ 2391.297799] hid-generic 0003:046D:C335.0010: input,hidraw0: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input0
[ 2391.298953] input: Logitech Gaming Keyboard G910 Keyboard as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.0011/input/input63
[ 2391.354559] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.0011/input/input64
[ 2391.354664] hid-generic 0003:046D:C335.0011: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input1
[ 2393.259237] usb 3-1.1: USB disconnect, device number 11
[ 2393.610886] usb 3-1.1: new full-speed USB device number 12 using xhci_hcd
[ 2393.718904] usb 3-1.1: New USB device found, idVendor=046d, idProduct=c335, bcdDevice= 6.02
[ 2393.718906] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2393.718907] usb 3-1.1: Product: Gaming Keyboard G910
[ 2393.718908] usb 3-1.1: Manufacturer: Logitech
[ 2393.718909] usb 3-1.1: SerialNumber: 0C6E384D3833
[ 2393.804293] input: Logitech Gaming Keyboard G910 as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.0/0003:046D:C335.0012/input/input67
[ 2393.860998] hid-generic 0003:046D:C335.0012: input,hidraw0: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input0
[... 48 lines cut ...]
[ 2422.010031] audit: type=1334 audit(1583166773.675:148): prog-id=15 op=UNLOAD
[ 2423.467246] usb 3-1.1: USB disconnect, device number 15
[ 2423.815627] usb 3-1.1: new full-speed USB device number 16 using xhci_hcd
[ 2423.927020] usb 3-1.1: New USB device found, idVendor=046d, idProduct=c335, bcdDevice= 6.02
[ 2423.927022] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2423.927024] usb 3-1.1: Product: Gaming Keyboard G910
[ 2423.927025] usb 3-1.1: Manufacturer: Logitech
[ 2423.927026] usb 3-1.1: SerialNumber: 0C6E384D3833
[... cut ...]
[ 2426.626923] input: Logitech Gaming Keyboard G910 Keyboard as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.001D/input/input93
[ 2426.682247] input: Logitech Gaming Keyboard G910 Consumer Control as /devices/pci0000:00/0000:00:07.1/0000:0c:00.3/usb3/3-1/3-1.1/3-1.1:1.1/0003:046D:C335.001D/input/input94
[ 2426.682407] hid-generic 0003:046D:C335.001D: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G910] on usb-0000:0c:00.3-1.1/input1

MatMoul, 2020-04-28:
I don't want to read all these lines...
Firstly, 
What's your OS and version?
What's your g810-led version?
You do you have installed g810-led?

jgslade, 2020-04-28:
I am using Manjaro, I installed it from the AUR.

MatMoul, 2020-04-28:
Ok, I maintain the AUR package, it's installed same as me...
Your two lines of explain seem a systemd unit config issue...
I suggest you delete g810-led-git and see if you experience the same nonresponse problem on startup.

jgslade, 2020-04-28:
If I uninstall g810-led-git it does boot normally and it keeps the last profile that was loaded. Though we did last power for a few hours there either day and the profile reset to the default and I had to reinstall to change the profile again.

If I disable the systemd until it still had problems on reboot.

MatMoul, 2020-04-28:
Your case is very strange, g810-led is called by udev when you connect or boot your computer.
g810-led take normally ~ 100ms.
If it is run multiple time, it's an external call.

jgslade, 2020-04-28:
Very interesting...
I'll check my system but I don't recall doing any manual calls. 😕

MatMoul, 2020-04-28:
Some ideas :
- Use of a USB hub
- Bad USB connection

Your issue seem a connect/disconnect case.

jgslade, 2020-04-28:
It is plugged into a USB hub. I'll plug it into it's own port and see if that resolves it.

jgslade, 2020-04-28:
That did it. Plugging the keyboard into it's own slot on the computer let to a successful bit sequence.

jgslade, 2020-04-28:
Plugging the keyboard in directly to the computer and reinstalling g810-leds seems to have fixed the issue. I'll need to do a few more boots to make sure.

jgslade, 2020-05-01:
I've had several successful boots since plugging the keyboard directly into the PC rather than through a hub.

closed 2020-05-01 by jgslade

d43m0nhLInt3r, 2020-06-01:
same issue here. I use Manjaro Gnome and installed the AUR package. I dont use a hub and the issue happen on every usb port.