MatMoul/g810-led #215: G413 Carbon -a -g -k commands don't work

cristianosucci, 2020-03-05:
Hello!
First of all thank you for supporting Logitech keyboards on Linux.

I'm on Ubuntu 18.04.4. I ran the guide step by step, I had to go into safe mode to log in as root.

I checked if my keyboard was recognized:
```
➜  ~ g810-led --list-keyboards
Device: Logitech - G413 Carbon Mechanical Gaming Keyboard
	Vendor ID: 046d
	Product ID: c33a
	Serial Number: 168F38693231
```
```
➜  ~ g810-led --print-device  
Device: Logitech - G413 Carbon Mechanical Gaming Keyboard
	Vendor ID: 046d
	Product ID: c33a
	Serial Number: 168F38693231
```
When I tested the commands:

`g810-led -dv 046d -dp c33a -tuk 1 -a 000000`
`g810-led -dv 046d -dp c33a -tuk 2 -a 000000`
`g810-led -dv 046d -dp c33a -tuk 3 -a 000000`

all leds turned off.

At this point I checked the issue https://github.com/MatMoul/g810-led/issues/99 and I tried the two commands:

`g810-led -dv 046d -dp c33a -tuk 3 -fx color all 80`
`g810-led -dv 046d -dp c33a -tuk 3 -fx breathing all 80 0a`

and it works! But just these two. :(

I tried without success the commands:
`g810-led -p /etc/g810-led/profile` (nothing happen)
`g810-led -a 00ff00` (all the leds went out even after the system restarted, i had to turn off and turn on my pc to turn on the leds again)
`g810-led -k w 00ff00` (nothing happen)
`g810-led -g arrows 00ff00` (nothing happen)

What could be the problem with these commands?

Thanks!

justasno, 2020-04-14:
@cristianosucci  Did you have any progress with this?

cristianosucci, 2020-04-14:
> @cristianosucci Did you have any progress with this?

unfortunately no, do you have the same problem?

justasno, 2020-04-14:
Yes, it seems only a couple commands work.

MatMoul, 2020-04-28:
g413 is very limited...
Looking in the code, -a seem implemented but not -g or -k
This keyboard is very different and don't need commit.

aprovera, 2020-05-06:
I can confirm both -a and -fx breathing work correctly on my G413 Carbon, without any need of using the `-dv`, `-dp` and `-tuk` arguments.

Since it only has monochrome red LEDs, you only need to specify the first two digits of the color value (i.e. 00-FF). Not sure if this observation also applies to the G413 Silver with its white LEDs.

Another thing I've noticed is that the intensity setting chosen through software seems to be independent from the FN-F7 brightness setting on the keyboard itself. If I set it up with `g413-led -a 33`, for instance, I can then lower the brightness even more using the key combination. 

I have found no way of storing any settings to the keyboard itself: I've tried using -fx-store, but when I unplug the keyboard and then plug it back in it goes back to the default factory settings (solid color, full brightness). The brigthness setting chosen on the keyboard itself via FN-F7 doesn't appear to stick, either.

MatMoul, 2020-05-16:
Closing because seem working, you can reopen if you want.
@aprovera `-fx-store` isn't supported for the G413, it's for more recent keyboard...

closed 2020-05-16 by MatMoul