MatMoul/g810-led #223: G915 support

LindaLawton, 2020-04-30:
Hi just tried this with my new G915 and it appears its not supported.

# lsusb with bluetooth

me@me-ThinkPad-X1-Extreme-2nd:~$ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 004: ID 17ef:3082 Lenovo USB3.1 Hub             
Bus 004 Device 003: ID 17ef:307f Lenovo 
Bus 004 Device 002: ID 17ef:307f Lenovo USB3.1 Hub             
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 2109:8818 VIA Labs, Inc. 
Bus 003 Device 004: ID 046d:0a45 Logitech, Inc. 960 Headset
Bus 003 Device 009: ID 2109:8818 VIA Labs, Inc. 
Bus 003 Device 013: ID 046d:c541 Logitech, Inc. 
Bus 003 Device 008: ID 17ef:3083 Lenovo USB2.0 Hub             
Bus 003 Device 006: ID 17ef:3081 Lenovo 
Bus 003 Device 003: ID 17ef:3080 Lenovo 
Bus 003 Device 002: ID 17ef:3080 Lenovo USB2.0 Hub             
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 06cb:00bd Synaptics, Inc. 
Bus 001 Device 003: ID 04f2:b67c Chicony Electronics Co., Ltd Integrated Camera
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 8087:0029 Intel Corp. 
Bus 001 Device 005: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Now I tried with both the numbers from 

>Bus 003 Device 013: ID 046d:c541 Logitech, Inc. 
>Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver

I also tried plugging it in though the cable  instead of using the bluetooth reserved  and i got 

#lsusb without bluetooth

>Bus 003 Device 014: ID 046d:c33e Logitech, Inc. 

linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c541 -tuk 1 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c541 -tuk 2 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c541 -tuk 3 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c52b -tuk 1 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c52b -tuk 2 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c52b -tuk 3 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c33e -tuk 1 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c33e -tuk 2 -a 000000
Matching or compatible device not found !
linda@linda-ThinkPad-X1-Extreme-2nd:~$ g810-led -dv 046d -dp c33e -tuk 3 -a 000000
Matching or compatible device not found !

Seams to not be working.  any tips would be appreciated im really tired of the random light show this thing is coming with. 

# my system.

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan

MatMoul, 2020-05-08:
With release 0.4.1, try this with your keyboard connected by cable without hub or adaptater :
`sudo g810-led -dv 046d -dp c33e -tuk 4 -a 000000`
But following issue #198, it seem g810-led need some more code :(

closed 2020-05-16 by MatMoul

LindaLawton, 2020-05-20:
this issue is for g915 not g810

MatMoul, 2020-05-24:
@LindaLawton `g810-led` is the binary...
-dv -dp is the USB adress of the keyboard
and -tuk is the protocol (4 is the latest for the g815)