MatMoul/g810-led #226: Only effects work, not stationary colors

kostaskonkk, 2020-05-16:
I am using a g815-SE on Ubuntu 18.04, and version 0.4.1 and although the effect profiles work well, I cannot get the stationary colors to display.
When I run:
> g815-led -p all_blue
nothing happens.
Does anyone have any idea, what could be going wrong?

MatMoul, 2020-05-16:
> I am using a g815-SE

Seem not exist...

> g512-led -p all_blue

G512 or G815 or unexistant G815-SE?

Also, try to add in the title the keyboard model that you use...

kostaskonkk, 2020-05-17:
Sorry for the mistake in the original question. My keyboard is the Logitech - G512 SE. I compiled it with both usb libraries but there was no difference. The funny thing is that with the same keyboard and computer on an Arch Linux installation, all color profiles work fine.

MatMoul, 2020-05-17:
Firstly use only hidapi on Ubuntu.
Secondly, try with sudo, I suspect that on Ubuntu you don't have implemented the udev files :
https://github.com/MatMoul/g810-led/blob/master/udev/g810-led.rules

kostaskonkk, 2020-05-18:
Ok, I changed to hidapi.
I tried with sudo, no difference and I checked the /etc/udev/rules.d directory and the g810-led.rules files is in there. 
What I can't understand is why are the effects working and not the simple colorschemes, it seems very strange!

MatMoul, 2020-05-18:
Looking your previous post `g512-led -dv 046d -dp c342 -a ff0000`
Use `sudo g512-led -a ff0000` as sudoer user or
`g512-led -a ff0000` as root (try to be precise with your commands)
An other idea is that to use an USB hub or an adaptater #114

kostaskonkk, 2020-05-19:
I have tried both suggestions but they didn't work. Anyway, it seems that this problem is encountered only by me.
Thanks a lot for your help and the great work that you have done with this program.
I will take a closer look to it myself, when I get some time. :)

efficientbug, 2020-05-19:
I'm having the same issue. I have two simple profiles, one set to all black and another set to another solid color (they're both only `a RRGGBB` and `c`, just like the samples provided). 

However, my keyboard was black (my "default" profile), and running `sudo g512-led -p /etc/g810-led/colored` didn't work, so I tried to see what was wrong by running `sudo g512-led -a ff0000` and it didn't do anything either (I even tried running `sudo g512-led -c` afterwards, not sure what `commit` does here). Looking a bit further into it, running `sudo g512-led -fx color keys 00ffff` did the job. A few months back, when I first installed the software it worked perfectly as it was (setting with `a RRGGBB` on the shell or loading one of the profiles), but now it stopped working.

How can I fix this? How do I provide more information to help? My keyboard is the G512 SE and is properly recognized by `g512-led --print-device` as:

```
Device: Logitech - G512 SE
Vendor ID: 046d
Product ID: c342
```
Thanks!

kostaskonkk, 2020-05-19:
`sudo g512-led -fx color keys 00ffff`
Worked also for me! :)

MatMoul, 2020-05-19:
Thanks for feedback...
One line is missing in the code.

MatMoul, 2020-05-19:
Bug corrected in 0.4.2, need confirmation...

efficientbug, 2020-05-19:
Seems to be working fine now, @MatMoul! Thank you very much!

closed 2020-05-20 by kostaskonkk