MatMoul/g810-led #228: sometimes on boot or waking from sleep they leds cycle serveral times

d43m0nhLInt3r, 2020-06-01:
same issue here. I use Manjaro Gnome and installed the AUR package. I dont use a hub and the issue happen on every usb port. #213

temperrr, 2020-07-01:
I had the same issue. This fixed it for me:
Go to
/etc/udev/rules.d/g810-led.rules

remove all rules except the ones saying 
```
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c336", MODE="666" RUN+="/usr/bied/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c32b", MODE="666" RUN+="/usr/bin/**g910-led** -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", MODE="666" RUN+="/usr/bin/**g910-led** -p /etc/g810-led/profile"
```

Please note that you should remove the ones with "subsystems" and leave the ones with " subsystem". I removed all the ones that were not my model number. ( I use a G910) 

**Please note that I have no clue what I did, only that it worked...**

wraith3690, 2021-03-28:
I had the same issue with a g910 on linux mint 20.1 mate and I can confirm that removing everything from /etc/udev/rules.d/g810-led.rules except the following 

`ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c32b", MODE="666" RUN+="/usr/bin/g910-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", MODE="666" RUN+="/usr/bin/g910-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c339", MODE="666" RUN+="/usr/bin/gpro-led -p /etc/g810-led/profile"` 

shortens the issue of the colour cycling to a max of 3 cycles note my keyboard is unresponsive until the cycling ends maybe that is related somehow idk just being thorough also one line is different from temperrr's post the last one I assume this is due to being a different OS

ModProg, 2021-04-21:
This also helped with a similar issue I am having on reconnecting the keyboard.

ModProg, 2021-04-27:
BTW here is a workaround using a lockfile, for anyone interested.
Just replace the RUN statement with this:
```sh
#!/bin/sh

LOCKFILE=/tmp/g910.lock

if [ ! -f "$LOCKFILE" ]; then
	touch $LOCKFILE

	/usr/bin/g910-led -p /etc/g810-led/profile

	(sleep 3;rm $LOCKFILE)&
fi
```

MPatzek, 2021-12-03:
Thanks! This solved the issue for me. Is there an updated repository somewhere that is including this fix?

ModProg, 2021-12-03:
No I haven't done anything in that direction. this also feels a bit like a hack but I don't have a better idea

MPatzek, 2021-12-03:
I just edited the files since I don't exactly know how to apply your Lockfile.
Could you explain shortly where to create this file?

ModProg, 2021-12-03:
> I just edited the files since I don't exactly know how to apply your Lockfile. Could you explain shortly where to create this file?

So did temperrr's work for you? Then you don't need to use the lockfile workaround.

If you need it, just place the script file anywhere and change the RUN+="..." to RUN+="/path/to/the/script.sh"

Panadur, 2022-01-27:
Using Debian Bullseye. 
Got the same issue and the workaround ModProg posted seems to slove the issue. Before the "patch" i got the issue on each reboot resulting in the keyboard beein not able to work for minutes sometimes until the cycling stopped.
thanks